f4pga/f4pga-arch-defs #1246: Create a tool to extract the library definitions from Xilinx

rodrigomelo9, 2019-12-19:
Hi @mithro this issue is to address your comment [here](https://github.com/YosysHQ/yosys/issues/1583#issuecomment-567158223) about the need for something similar to xc7/library for Spartan 6. I read #149 and check what was done in xc7/libraries.

I want to know:
* What is exactly needed? I know, the module name, the ports name, size and directions and the order, but what is your input there? The xml file? The generated Verilog file that is not there? Could I know what is exactly the idea?
* Was proposed to be extracted for the PDF because...? There are Verilog files for each primitive under `<VIVADO_ROOT_PATH>/data/verilog/src/unisims` (and other places, but it seems to me that the most updated versions are there). Maybe, we could use the PDFs to extract only the names of the primitives per family, and after that, get data from there? Also, I saw at least with series 7 in Vivado, that we can also use info from `data/parts/xilinx/<FAMILY>/devint/<FAMILY>.lib`.

What do you think? Were these other resources evaluated and discarded? The PDF option was only an idea or was chosen after the evaluation of the best option?

Let me say that I like this kind of task, so I am interesting in collect from the vendor tools what could be useful for the project.

mithro, 2019-12-19:
The stuff here is being done on the advice from our legal experts.

We can **not** include code from Vivado in the SymbiFlow project as they are not under a compatible open source license.

Extracting the **minimal** information needed for compatibility with Vivado or ISE from the PDF documentation was determined to be okay.

rodrigomelo9, 2019-12-19:
I got it.

mithro, 2019-12-19:
I think the approach was to extract the PDF information into an XML file. Then from the XML file we could generate the required Verilog and VHDL black box modules (and potentially other things like diagrams using https://kevinpt.github.io/symbolator/ or https://wavedrom.com/).

rodrigomelo9, 2019-12-20:
Here we have the same problem as in #1243: not all the primitives have the Port Description info available in a table. There are instantiation examples... but the ports are not necessarily in order (I saw cases where the order in the Port desc table is not the same that in the example instantiation). Is the port order in the documentation really important @mithro ? (you said that the order is needed in #149).

mithro, 2019-12-20:
@mithro - I believe order is important because you can specify the ports in Verilog via order?

rodrigomelo9, 2019-12-20:
ahh ok, yes, when is assigned by position (I always avoided this practice but you right). It seems that definitively we will need patches...

rodrigomelo9, 2019-12-20:
@mjasperse I used your last commit on `parse_pdf_modules.py` with UG615 (v14.7 - October 2, 2013), and it works :-D the only warnig of the tool was:
```
Processing DCM_CLKGEN...
Warning: port STATUS[2:1] does not match pattern
```
It gives me 100 modules when I count 101 manually XD but is near :P with the same problem about no *port desc*.