f4pga/f4pga-arch-defs #1250: Missing expected argument for --router_lookahead

mithro, 2019-12-20:
Looks like there is a failure on the latest commit of https://github.com/SymbiFlow/symbiflow-arch-defs/commit/6575fce443854c489f621e1f7e86da3f5e562494

See the failure at https://source.cloud.google.com/results/invocations/e8b064d0-ea51-447f-a1e5-631a71dc706c/log
```
Missing expected argument for --router_lookahead

usage: vpr architecture circuit [--pack] [--place] [--route] [--analysis]
       [--disp {on, off}] [--save_graphics {on, off}] [-h] [--version]
       [--device DEVICE_NAME] [-j NUM_WORKERS] [--timing_analysis {on, off}]
       [--disable_errors DISABLE_ERRORS] [--suppress_warnings SUPPRESS_WARNINGS]
       [--route_chan_width CHANNEL_WIDTH] [OTHER_OPTIONS ...]
```

mithro, 2019-12-20:
Something weird is going on there is a `--router_lookahead connection_box_map` in the output...
```
[939/1912] Generating rr_graph_xc7a50t-basys3_test.rr_graph.real.xml.cache, rr_graph_xc7a50t-basys3_test.lookahead.bin, rr_graph_xc7a50t-basys3_test.place_delay.bin
FAILED: cd /tmpfs/src/github/symbiflow-arch-defs/build/xc7/archs/artix7/devices && /tmpfs/src/github/symbiflow-arch-defs/build/env/conda/bin/python3 /tmpfs/src/github/symbiflow-arch-defs/utils/check_cache.py /tmpfs/src/github/symbiflow-arch-defs/build/xc7/archs/artix7/devices/rr_graph_xc7a50t-basys3_test.rr_graph.real.xml /tmpfs/src/github/symbiflow-arch-defs/build/xc7/archs/artix7/devices/rr_graph_xc7a50t-basys3_test.rr_graph.real.xml.cache rr_graph_xc7a50t-basys3_test.lookahead.bin rr_graph_xc7a50t-basys3_test.place_delay.bin || ( /tmpfs/src/github/symbiflow-arch-defs/utils/quiet_cmd.sh /tmpfs/src/github/vtr-verilog-to-routing/build/vpr/vpr /tmpfs/src/github/symbiflow-arch-defs/build/xc7/archs/artix7/devices/xc7a50t-basys3-roi-virt/arch.timing.xml --device xc7a50t-basys3-test /tmpfs/src/github/symbiflow-arch-defs/common/wire.eblif --read_rr_graph /tmpfs/src/github/symbiflow-arch-defs/build/xc7/archs/artix7/devices/rr_graph_xc7a50t-basys3_test.rr_graph.real.xml --outfile_prefix xc7a50t-basys3_test_cache --pack --place --write_router_lookahead rr_graph_xc7a50t-basys3_test.lookahead.bin --write_placement_delay_lookup rr_graph_xc7a50t-basys3_test.place_delay.bin --constant_net_method route --clock_modeling route --place_delay_model delta_override --router_lookahead connection_box_map --disable_errors check_unbuffered_edges:check_route:check_place --suppress_warnings sum_pin_class:check_unbuffered_edges:load_rr_indexed_data_T_values:check_rr_node:trans_per_R --route_chan_width 500 --allowed_tiles_for_delay_model BLK-TL-SLICEL,BLK-TL-SLICEM && /tmpfs/src/github/symbiflow-arch-defs/build/env/conda/bin/python3 /tmpfs/src/github/symbiflow-arch-defs/utils/update_cache.py /tmpfs/src/github/symbiflow-arch-defs/build/xc7/archs/artix7/devices/rr_graph_xc7a50t-basys3_test.rr_graph.real.xml /tmpfs/src/github/symbiflow-arch-defs/build/xc7/archs/artix7/devices/rr_graph_xc7a50t-basys3_test.rr_graph.real.xml.cache ) && /usr/bin/cmake -E copy vpr_stdout.log rr_graph_xc7a50t-basys3_test.cache.out
```

mithro, 2019-12-20:
Is it related to https://github.com/SymbiFlow/symbiflow-arch-defs/issues/1187 ?

mithro, 2019-12-20:
Is this the right VPR version?
```
Version: 8.0.0-rc1+2780988d5
Revision: v8.0.0-rc1-1212-g2780988d5
Compiled: 2019-12-20T06:39:59
Compiler: GNU 5.4.0 on Linux-4.4.0-116-generic x86_64
Build Info: release IPO VTR_ASSERT_LEVEL=3
```

litghost, 2020-01-02:
This bug is because the kokoro CI is using the master branch of Symbiflow/vtr rather than the master+wip branch, and as a result it errors out because the modifications required to run the symbiflow-arch-def flow are missing.  It is unclear if this was the intention when this target was set up.

There is another kokoro configuration that uses Symbiflow/vtr master+wip, and that one works as expected.

litghost, 2020-10-29:
This will be fixed once upstream VPR can run with symbiflow-arch-defs, which is soon!

closed 2020-10-29 by litghost

syed-ahmed, 2020-12-04:
Is this fixed? I'm seeing this error when using `--router_lookahead connection_box_map` on current master.

```
VPR FPGA Placement and Routing.
Version: 8.1.0-dev+d6d69ff92
Revision: v8.0.0.rc2-5415-gd6d69ff92
Compiled: 2020-11-20T18:17:03
Compiler: GNU 7.3.0 on Linux-4.15.0-1077-gcp x86_64
Build Info: Release IPO VTR_ASSERT_LEVEL=2

University of Toronto
verilogtorouting.org
vtr-users@googlegroups.com
This is free open source code under MIT license.

Missing expected argument for --router_lookahead

usage: vpr architecture circuit [--pack] [--place] [--route] [--analysis] 
       [--disp {on, off}] [--save_graphics {on, off}] 
       [--graphics_commands GRAPHICS_COMMANDS] [-h] [--version] 
       [--device DEVICE_NAME] [-j NUM_WORKERS] [--timing_analysis {on, off}] 
       [--disable_errors DISABLE_ERRORS] [--suppress_warnings SUPPRESS_WARNINGS] 
       [--route_chan_width CHANNEL_WIDTH] [OTHER_OPTIONS ...]
make: [Makefile:77: output_fun/out.net] Error 1 (ignored)
```
I'm using vpr from: `symbiflow-arch-defs/env/conda/envs/symbiflow_arch_def_base/bin/vpr`. Any help is appreciated!

litghost, 2020-12-04:
It's fixed in the sense that we no longer used `connection_box_map` at all?  We've actually just removed it from our build of VPR, and are currently using upstream VPR directly.

litghost, 2020-12-04:
PR with relevant change: https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1667

syed-ahmed, 2020-12-04:
Thanks! That worked.