f4pga/f4pga-arch-defs #1258: Incorrect handling of OSERDES inputs in the techmap

mkurc-ant, 2020-01-03:
While trying to make fasm2bels work on the minilitex_ddr design (https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1220) I discovered the following behavior of vendor tools regarding OSERDES inputs connected to constants:

For data ports D1-D8:

|Connection|Routed to|fasm output|
|---|---|---|
|Unconnected|<const 1>|IS_Dx_INVERTED|
|0|<const 1>|IS_Dx_INVERTED|
|1|<const 1>|-|
|net|net|-|

For tri-state control ports T1-T4:

|Connection|Routed to|fasm output|
|---|---|---|
|Unconnected|<const 1>|-|
|0|<const 1>|ZINV_Tx|
|1|<const 1>|-|
|net|net|ZINV_Tx|

It seems that vendor tools never route `<const 0>` to the OSERDES.

Similar behavior can be observed for `OCE` and `TCE` ports. They are routed to `<const 1>` when unconnected (I didn't check inverters though).

Setting/clearing the parameter `IS_Dx_INVERTED` or `IS_Tx_INVERTED` of an `OSERDES` cell additionally toggles the presence/absence of the emitted fasm feature. So the feature value needs to be xored with the parameter prior to emission.

The feature `IS_Dx_INVERTED` enables the inverter on the data line but the feature `ZINV_Tx` actually DISABLES the inverter on the tri-state input.

mkurc-ant, 2020-01-03:
@acomodi FYI

litghost, 2020-01-03:
@mkurc-ant can you please create a smaller failing test case in xc7/tests/serdes?  This behavior is typical, I'm surprised it wasn't caught earlier.

litghost, 2020-01-03:
For both @mkurc-ant and @acomodi reference, all input pins are connected to `<const1>` when no bits are set in the bitstream.  Inverters are typically set such when the invert feature is **not** set, the hardware is disabled.  As a concrete example, the enable port of the BUFGCTRL defaults to `<const0>` because the inverter on the control line defaults to being **enabled**.  So the default routing fabric routes a `<const1>` to the inverter, the inverter inverts by default, resulting in a `<const0>` being sent to the enable line, disabling the hardware.

Does that make sense?

mkurc-ant, 2020-01-03:
@litghost Yes, all of that makes sense to me.

My findings that I described in this issue are based on a problem encountered with fasm2bels ran on the LiteX with DDR design. When eg. all T inputs of an OSERDES are connected to 0 (explicitly) they are routed to <const 0> by SymbiFlow with inverters disabled. However, when importing design decoded by fasm2bels to Vivado it complains about routing conflict. Vivado expects all those inputs to be routed to <const 1> and have inverters enabled.

I have a tiny test that I synthesized with Vivado and observed how the routing + inverters behave. I can port it to use with SymbiFlow.

litghost, 2020-01-03:
> @litghost Yes, all of that makes sense to me.
>
> My findings that I described in this issue are based on a problem encountered with fasm2bels ran on the LiteX with DDR design. When eg. all T inputs of [...]

That pattern is common in the 7-series fabric.  Take a look at the bram techmap, it has logic to handle exactly what you are describing.

mkurc-ant, 2020-01-08:
I've added a test demonstrating this issue: https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1261