f4pga/f4pga-arch-defs #1269: Vendor tools CI failing on DRAM tests

acomodi, 2020-01-17:
Vendor tools dram tests were not enabled until #1234 got merged.

With https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1268 I have temporarily disabled the vivado_targets, to let CI go green (as it has been red for too long now).

This issue is to keep track of the problem with DRAM evaluated on vendor tools with fasm2bels.

litghost, 2020-01-23:
The current LUT-RAM implementation is likely wrong after #1267, but less wrong than before.

In particular:
 - RAM32X1S should be able to pack between 6-8 BELs in a SLICEM, currently only 4 are accomidated

For future work around this:
 - [ ] We need to consult the relevant documentation, and determine what are the valid packing schemes for 32-bit LUT-RAMs
 - [ ] Develop a set of Vivado minitests demostrating the various packing options
 - [ ] Add those to symbiflow-arch-defs and verify that VPR can pack that tightly
 - [ ] Verify that fasm2bels can round trip those designs for analysis
 - [ ] Re-enable 32-bit LUT-RAM synthesis once previous steps are completed (e.g. https://github.com/SymbiFlow/yosys/pull/31)

litghost, 2020-01-23:
I've split the 32-bit LUT-RAM issues and 128/256 bit LUT-RAM issues into two seperate issues, with the 128/256 issues into #1275.  This is because the fixes around the 128/256 is likely much less work than the fixes around the 32-bit LUT-RAM issues.