f4pga/f4pga-arch-defs #1274: ReadTheDocs CI target randomly failing

acomodi, 2020-01-23:
In some cases, the ReadTheDocs CI build fails with the following message:

```
Traceback (most recent call last):
  File "/home/docs/.pyenv/versions/2.7.16/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/home/docs/.pyenv/versions/2.7.16/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/docs/checkouts/readthedocs.org/user_builds/symbiflow-arch-defs/envs/1267/lib/python2.7/site-packages/pip/__main__.py", line 16, in <module>
    from pip._internal import main as _main  # isort:skip # noqa
  File "/home/docs/checkouts/readthedocs.org/user_builds/symbiflow-arch-defs/envs/1267/lib/python2.7/site-packages/pip/_internal/__init__.py", line 40, in <module>
    from pip._internal.cli.autocompletion import autocomplete
  File "/home/docs/checkouts/readthedocs.org/user_builds/symbiflow-arch-defs/envs/1267/lib/python2.7/site-packages/pip/_internal/cli/autocompletion.py", line 8, in <module>
    from pip._internal.cli.main_parser import create_main_parser
  File "/home/docs/checkouts/readthedocs.org/user_builds/symbiflow-arch-defs/envs/1267/lib/python2.7/site-packages/pip/_internal/cli/main_parser.py", line 12, in <module>
    from pip._internal.commands import (
  File "/home/docs/checkouts/readthedocs.org/user_builds/symbiflow-arch-defs/envs/1267/lib/python2.7/site-packages/pip/_internal/commands/__init__.py", line 6, in <module>
    from pip._internal.commands.completion import CompletionCommand
  File "/home/docs/checkouts/readthedocs.org/user_builds/symbiflow-arch-defs/envs/1267/lib/python2.7/site-packages/pip/_internal/commands/completion.py", line 6, in <module>
    from pip._internal.cli.base_command import Command
  File "/home/docs/checkouts/readthedocs.org/user_builds/symbiflow-arch-defs/envs/1267/lib/python2.7/site-packages/pip/_internal/cli/base_command.py", line 25, in <module>
    from pip._internal.index import PackageFinder
ImportError: cannot import name PackageFinder
```

mithro, 2020-01-23:
This is weird as it shouldn't be using Python 2.7?

umarcor, 2022-04-06:
A RTD config file was added a few months ago (https://github.com/SymbiFlow/f4pga-arch-defs/commit/1cfd9f52c334ba5e1fb1fbc118860d399b524ae0) which uses Python 3.7 explecitily. Hence, I'm closing this.

closed 2022-04-06 by umarcor