f4pga/f4pga-arch-defs #1278: Combinatorial Loop after running fasm2bels

acomodi, 2020-01-24:
When experimenting on the `antenna nets` generated with some designs for the artix50T, I encountered this issue after having rerouted the faulty antenna nets with Vivado (the design is `murax_basys_full`):

```
ERROR: [DRC LUTLP-1] Combinatorial Loop Alert: 1 LUT cells form a combinatorial loop. This can create a race condition. Timing analysis may not be accurate. The preferred resolution is to modify the design to remove combinatorial logic loops. If the loop is known and understood, this DRC can be bypassed by acknowledging the condition and setting the following XDC constraint on any net in the loop: 'set_property ALLOW_COMBINATORIAL_LOOPS TRUE [net_nets <myHier/myNet>'. The cells in the loop are: CLBLL_L_X26Y105_SLICE_X40Y105_CLUT/LUT6.
ERROR: [DRC LUTLP-1] Combinatorial Loop Alert: 1 LUT cells form a combinatorial loop. This can create a race condition. Timing analysis may not be accurate. The preferred resolution is to modify the design to remove combinatorial logic loops. If the loop is known and understood, this DRC can be bypassed by acknowledging the condition and setting the following XDC constraint on any net in the loop: 'set_property ALLOW_COMBINATORIAL_LOOPS TRUE [net_nets <myHier/myNet>'. The cells in the loop are: CLBLL_L_X28Y91_SLICE_X44Y91_BLUT/LUT6.
ERROR: [DRC LUTLP-1] Combinatorial Loop Alert: 1 LUT cells form a combinatorial loop. This can create a race condition. Timing analysis may not be accurate. The preferred resolution is to modify the design to remove combinatorial logic loops. If the loop is known and understood, this DRC can be bypassed by acknowledging the condition and setting the following XDC constraint on any net in the loop: 'set_property ALLOW_COMBINATORIAL_LOOPS TRUE [net_nets <myHier/myNet>'. The cells in the loop are: CLBLL_R_X15Y107_SLICE_X23Y107_DLUT/LUT6.
ERROR: [DRC LUTLP-1] Combinatorial Loop Alert: 1 LUT cells form a combinatorial loop. This can create a race condition. Timing analysis may not be accurate. The preferred resolution is to modify the design to remove combinatorial logic loops. If the loop is known and understood, this DRC can be bypassed by acknowledging the condition and setting the following XDC constraint on any net in the loop: 'set_property ALLOW_COMBINATORIAL_LOOPS TRUE [net_nets <myHier/myNet>'. The cells in the loop are: CLBLL_R_X31Y22_SLICE_X48Y22_BLUT/LUT6.
ERROR: [DRC LUTLP-1] Combinatorial Loop Alert: 1 LUT cells form a combinatorial loop. This can create a race condition. Timing analysis may not be accurate. The preferred resolution is to modify the design to remove combinatorial logic loops. If the loop is known and understood, this DRC can be bypassed by acknowledging the condition and setting the following XDC constraint on any net in the loop: 'set_property ALLOW_COMBINATORIAL_LOOPS TRUE [net_nets <myHier/myNet>'. The cells in the loop are: CLBLM_L_X20Y77_SLICE_X30Y77_BLUT/LUT6.
ERROR: [DRC LUTLP-1] Combinatorial Loop Alert: 1 LUT cells form a combinatorial loop. This can create a race condition. Timing analysis may not be accurate. The preferred resolution is to modify the design to remove combinatorial logic loops. If the loop is known and understood, this DRC can be bypassed by acknowledging the condition and setting the following XDC constraint on any net in the loop: 'set_property ALLOW_COMBINATORIAL_LOOPS TRUE [net_nets <myHier/myNet>'. The cells in the loop are: CLBLM_L_X22Y67_SLICE_X34Y67_ALUT/LUT6.
ERROR: [DRC LUTLP-1] Combinatorial Loop Alert: 1 LUT cells form a combinatorial loop. This can create a race condition. Timing analysis may not be accurate. The preferred resolution is to modify the design to remove combinatorial logic loops. If the loop is known and understood, this DRC can be bypassed by acknowledging the condition and setting the following XDC constraint on any net in the loop: 'set_property ALLOW_COMBINATORIAL_LOOPS TRUE [net_nets <myHier/myNet>'. The cells in the loop are: CLBLM_R_X25Y103_SLICE_X38Y103_ALUT/LUT6.
ERROR: [DRC LUTLP-1] Combinatorial Loop Alert: 1 LUT cells form a combinatorial loop. This can create a race condition. Timing analysis may not be accurate. The preferred resolution is to modify the design to remove combinatorial logic loops. If the loop is known and understood, this DRC can be bypassed by acknowledging the condition and setting the following XDC constraint on any net in the loop: 'set_property ALLOW_COMBINATORIAL_LOOPS TRUE [net_nets <myHier/myNet>'. The cells in the loop are: CLBLM_R_X25Y103_SLICE_X38Y103_BLUT/LUT6.
ERROR: [DRC LUTLP-1] Combinatorial Loop Alert: 1 LUT cells form a combinatorial loop. This can create a race condition. Timing analysis may not be accurate. The preferred resolution is to modify the design to remove combinatorial logic loops. If the loop is known and understood, this DRC can be bypassed by acknowledging the condition and setting the following XDC constraint on any net in the loop: 'set_property ALLOW_COMBINATORIAL_LOOPS TRUE [net_nets <myHier/myNet>'. The cells in the loop are: CLBLM_R_X27Y31_SLICE_X42Y31_BLUT/LUT6.
ERROR: [DRC LUTLP-1] Combinatorial Loop Alert: 1 LUT cells form a combinatorial loop. This can create a race condition. Timing analysis may not be accurate. The preferred resolution is to modify the design to remove combinatorial logic loops. If the loop is known and understood, this DRC can be bypassed by acknowledging the condition and setting the following XDC constraint on any net in the loop: 'set_property ALLOW_COMBINATORIAL_LOOPS TRUE [net_nets <myHier/myNet>'. The cells in the loop are: CLBLM_R_X35Y53_SLICE_X54Y53_BLUT/LUT6.
ERROR: [DRC LUTLP-1] Combinatorial Loop Alert: 2 LUT cells form a combinatorial loop. This can create a race condition. Timing analysis may not be accurate. The preferred resolution is to modify the design to remove combinatorial logic loops. If the loop is known and understood, this DRC can be bypassed by acknowledging the condition and setting the following XDC constraint on any net in the loop: 'set_property ALLOW_COMBINATORIAL_LOOPS TRUE [net_nets <myHier/myNet>'. The cells in the loop are: CLBLL_L_X26Y105_SLICE_X40Y105_CLUT/LUT5, and CLBLL_L_X26Y105_SLICE_X40Y105_CLUT/LUT6.
ERROR: [DRC LUTLP-1] Combinatorial Loop Alert: 2 LUT cells form a combinatorial loop. This can create a race condition. Timing analysis may not be accurate. The preferred resolution is to modify the design to remove combinatorial logic loops. If the loop is known and understood, this DRC can be bypassed by acknowledging the condition and setting the following XDC constraint on any net in the loop: 'set_property ALLOW_COMBINATORIAL_LOOPS TRUE [net_nets <myHier/myNet>'. The cells in the loop are: CLBLL_L_X2Y77_SLICE_X0Y77_ALUT/LUT6, and CLBLL_L_X2Y77_SLICE_X0Y77_CLUT/LUT6.
ERROR: [DRC LUTLP-1] Combinatorial Loop Alert: 3 LUT cells form a combinatorial loop. This can create a race condition. Timing analysis may not be accurate. The preferred resolution is to modify the design to remove combinatorial logic loops. If the loop is known and understood, this DRC can be bypassed by acknowledging the condition and setting the following XDC constraint on any net in the loop: 'set_property ALLOW_COMBINATORIAL_LOOPS TRUE [net_nets <myHier/myNet>'. The cells in the loop are: CLBLM_L_X22Y75_SLICE_X34Y75_DLUT/LUT5, CLBLM_L_X22Y75_SLICE_X34Y75_ALUT/LUT6, and CLBLM_L_X22Y75_SLICE_X34Y75_CLUT/LUT6.
ERROR: [DRC LUTLP-1] Combinatorial Loop Alert: 4 LUT cells form a combinatorial loop. This can create a race condition. Timing analysis may not be accurate. The preferred resolution is to modify the design to remove combinatorial logic loops. If the loop is known and understood, this DRC can be bypassed by acknowledging the condition and setting the following XDC constraint on any net in the loop: 'set_property ALLOW_COMBINATORIAL_LOOPS TRUE [net_nets <myHier/myNet>'. The cells in the loop are: CLBLL_L_X24Y92_SLICE_X36Y92_DLUT/LUT5, CLBLL_L_X24Y92_SLICE_X36Y92_CLUT/LUT6, CLBLL_L_X26Y59_SLICE_X41Y59_BLUT/LUT6, and CLBLL_L_X26Y59_SLICE_X41Y59_DLUT/LUT6.
ERROR: [DRC LUTLP-1] Combinatorial Loop Alert: 4 LUT cells form a combinatorial loop. This can create a race condition. Timing analysis may not be accurate. The preferred resolution is to modify the design to remove combinatorial logic loops. If the loop is known and understood, this DRC can be bypassed by acknowledging the condition and setting the following XDC constraint on any net in the loop: 'set_property ALLOW_COMBINATORIAL_LOOPS TRUE  [... cut ...]  <myHier/myNet>'. The cells in the loop are: CLBLL_L_X28Y57_SLICE_X44Y57_BLUT/LUT6, CLBLL_L_X28Y57_SLICE_X44Y57_CLUT/LUT6, CLBLL_R_X19Y63_SLICE_X28Y63_ALUT/LUT6, CLBLL_R_X19Y63_SLICE_X28Y63_BLUT/LUT6, CLBLM_L_X20Y70_SLICE_X30Y70_BLUT/LUT6, CLBLM_L_X20Y70_SLICE_X30Y70_DLUT/LUT6, CLBLM_L_X22Y92_SLICE_X35Y92_DLUT/LUT6, and CLBLM_R_X25Y111_SLICE_X38Y111_CLUT/LUT6.

```

I am unsure whether this is relative to the rerouting of the antenna nets with Vivado, or the issue is generated withing SymbiFlow.

litghost, 2020-01-24:
FYI the tcl file generated for fasm2bels disable those loops: https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc7/utils/vivado_create_runme.py#L48

As for whether these loops are real, I believe they are not.  In particular, fasm2bels shows higher LUT usage than a real design because of how it emits to verilog.  I believe that the problem is that Vivado is not verifying if the LUT equations form a loop.  If LUT A . O6 forms a loop with LUT B, but via an address line that doesn't affect O6, is that a Combinatorial loop?

mithro, 2020-01-29:
Is this a case of a LUT6_2 where the O6 and O5 outputs are dependent only
on a certain set of inputs? (IE can be thought of as two independent LUTs
packed into the one primitive?)

litghost, 2020-01-29:
> Is this a case of a LUT6_2 where the O6 and O5 outputs are dependent only on a certain set of inputs? (IE can be thought of as two independent LUTs packed into the one primitive?)

That is my assumption.