f4pga/f4pga-arch-defs #1291: SDC/XDC create_clock and set_false_path constraints should propigate through Yosys

litghost, 2020-01-31:
Currently SDC constraints are passed directly to VPR, which means the net names that are output from Yosys need to be used if the constraints are for nets, rather than top level ports.  Constraints should be specified on the nets in the input verilog, and then a new SDC constraints file should be output from Yosys with the output net names.

@tmichalak / @acomodi I believe this would allow the LiteX XDC constraints to be used throughout the flow?

@mithro FYI

mithro, 2020-01-31:
Yes.

We can also do the clock constraint propagation through PLLs and and other boxes here.

mithro, 2020-02-01:
@daveshah1 - FYI this is the direction we are planning on going with constraints with VPR. Do you think it would be useful for nextpnr at all?

daveshah1, 2020-02-01:
I implemented net aliases in nextpnr, so a constraint in nextpnr can be any of the names that Yosys gives to a net. But this could be useful too

mithro, 2020-02-01:
@daveshah1 How do you figure out the net aliases?

daveshah1, 2020-02-01:
In the JSON, the netnames section format allows more than one name for each net. I don't know if there is a BLIF equivalent.

litghost, 2020-10-29:
Can you close this issue once this is merged?

tmichalak, 2020-11-02:
`create_clock` propagates though Yosys with this [SDC plugin PR](https://github.com/SymbiFlow/yosys-symbiflow-plugins/pull/27) and a [fix](https://github.com/SymbiFlow/yosys-symbiflow-plugins/pull/51).
`set_false_path` is not propagated, currently we simply write out what we get from the commands.

tmichalak, 2020-12-04:
All related PRs have been merged. Closing.

mithro, 2020-12-05:
Is there a separate issue so that `set_false_path` is handled correctly?