f4pga/f4pga-arch-defs #1307: `minilitex` diff FASM test shows unknown bits

litghost, 2020-02-12:
It turns out the `_diff_fasm` tests were not running `bit2fasm` with the `--verbose` flag, meaning that unknown bits generated from Vivado were not reported.  When `--verbose` was added, the result was that the `minilitex` design reported unknown bits:

```
+# In frame 0x00000f1b 32 bits were not converted.
+{ unknown_bit = "00000f1b_5_19", unknown_segment = "0x00000f00", unknown_segbit = "27_179" }
+{ unknown_bit = "00000f1b_5_20", unknown_segment = "0x00000f00", unknown_segbit = "27_180" }
+{ unknown_bit = "00000f1b_5_21", unknown_segment = "0x00000f00", unknown_segbit = "27_181" }
+{ unknown_bit = "00000f1b_5_24", unknown_segment = "0x00000f00", unknown_segbit = "27_184" }
+{ unknown_bit = "00000f1b_15_19", unknown_segment = "0x00000f00", unknown_segbit = "27_499" }
+{ unknown_bit = "00000f1b_15_20", unknown_segment = "0x00000f00", unknown_segbit = "27_500" }
+{ unknown_bit = "00000f1b_15_21", unknown_segment = "0x00000f00", unknown_segbit = "27_501" }
+{ unknown_bit = "00000f1b_15_24", unknown_segment = "0x00000f00", unknown_segbit = "27_504" }
+{ unknown_bit = "00000f1b_25_19", unknown_segment = "0x00000f00", unknown_segbit = "27_819" }
+{ unknown_bit = "00000f1b_25_20", unknown_segment = "0x00000f00", unknown_segbit = "27_820" }
+{ unknown_bit = "00000f1b_25_21", unknown_segment = "0x00000f00", unknown_segbit = "27_821" }
+{ unknown_bit = "00000f1b_25_24", unknown_segment = "0x00000f00", unknown_segbit = "27_824" }
+{ unknown_bit = "00000f1b_45_19", unknown_segment = "0x00000f00", unknown_segbit = "27_1459" }
+{ unknown_bit = "00000f1b_45_20", unknown_segment = "0x00000f00", unknown_segbit = "27_1460" }
+{ unknown_bit = "00000f1b_45_21", unknown_segment = "0x00000f00", unknown_segbit = "27_1461" }
+{ unknown_bit = "00000f1b_45_24", unknown_segment = "0x00000f00", unknown_segbit = "27_1464" }
+{ unknown_bit = "00000f1b_66_19", unknown_segment = "0x00000f00", unknown_segbit = "27_2131" }
+{ unknown_bit = "00000f1b_66_20", unknown_segment = "0x00000f00", unknown_segbit = "27_2132" }
+{ unknown_bit = "00000f1b_66_21", unknown_segment = "0x00000f00", unknown_segbit = "27_2133" }
+{ unknown_bit = "00000f1b_66_24", unknown_segment = "0x00000f00", unknown_segbit = "27_2136" }
+{ unknown_bit = "00000f1b_76_19", unknown_segment = "0x00000f00", unknown_segbit = "27_2451" }
+{ unknown_bit = "00000f1b_76_20", unknown_segment = "0x00000f00", unknown_segbit = "27_2452" }
+{ unknown_bit = "00000f1b_76_21", unknown_segment = "0x00000f00", unknown_segbit = "27_2453" }
+{ unknown_bit = "00000f1b_76_24", unknown_segment = "0x00000f00", unknown_segbit = "27_2456" }
+{ unknown_bit = "00000f1b_86_19", unknown_segment = "0x00000f00", unknown_segbit = "27_2771" }
+{ unknown_bit = "00000f1b_86_20", unknown_segment = "0x00000f00", unknown_segbit = "27_2772" }
+{ unknown_bit = "00000f1b_86_21", unknown_segment = "0x00000f00", unknown_segbit = "27_2773" }
+{ unknown_bit = "00000f1b_86_24", unknown_segment = "0x00000f00", unknown_segbit = "27_2776" }
+{ unknown_bit = "00000f1b_96_19", unknown_segment = "0x00000f00", unknown_segbit = "27_3091" }
+{ unknown_bit = "00000f1b_96_20", unknown_segment = "0x00000f00", unknown_segbit = "27_3092" }
+{ unknown_bit = "00000f1b_96_21", unknown_segment = "0x00000f00", unknown_segbit = "27_3093" }
+{ unknown_bit = "00000f1b_96_24", unknown_segment = "0x00000f00", unknown_segbit = "27_3096" }
+
+# In frame 0x00020f1b 8 bits were not converted.
+{ unknown_bit = "00020f1b_66_19", unknown_segment = "0x00020f00", unknown_segbit = "27_2131" }
+{ unknown_bit = "00020f1b_66_20", unknown_segment = "0x00020f00", unknown_segbit = "27_2132" }
+{ unknown_bit = "00020f1b_66_21", unknown_segment = "0x00020f00", unknown_segbit = "27_2133" }
+{ unknown_bit = "00020f1b_66_24", unknown_segment = "0x00020f00", unknown_segbit = "27_2136" }
+{ unknown_bit = "00020f1b_56_19", unknown_segment = "0x00020f00", unknown_segbit = "27_1811" }
+{ unknown_bit = "00020f1b_56_20", unknown_segment = "0x00020f00", unknown_segbit = "27_1812" }
+{ unknown_bit = "00020f1b_56_21", unknown_segment = "0x00020f00", unknown_segbit = "27_1813" }
+{ unknown_bit = "00020f1b_56_24", unknown_segment = "0x00020f00", unknown_segbit = "27_1816" }
+
+# In frame 0x00400f1b 24 bits were not converted.
+{ unknown_bit = "00400f1b_76_19", unknown_segment = "0x00400f00", unknown_segbit = "27_2451" }
+{ unknown_bit = "00400f1b_76_20", unknown_segment = "0x00400f00", unknown_segbit = "27_2452" }
+{ unknown_bit = "00400f1b_76_21", unknown_segment = "0x00400f00", unknown_segbit = "27_2453" }
+{ unknown_bit = "00400f1b_96_19", unknown_segment = "0x00400f00", unknown_segbit = "27_3091" }
+{ unknown_bit = "00400f1b_96_20", unknown_segment = "0x00400f00", unknown_segbit = "27_3092" }
+{ unknown_bit = "00400f1b_76_24", unknown_segment = "0x00400f00", unknown_segbit = "27_2456" }
+{ unknown_bit = "00400f1b_96_21", unknown_segment = "0x00400f00", unknown_segbit = "27_3093" }
+{ unknown_bit = "00400f1b_96_24", unknown_segment = "0x00400f00", unknown_segbit = "27_3096" }
+{ unknown_bit = "00400f1b_56_19", unknown_segment = "0x00400f00", unknown_segbit = "27_1811" }
+{ unknown_bit = "00400f1b_56_20", unknown_segment = "0x00400f00", unknown_segbit = "27_1812" }
+{ unknown_bit = "00400f1b_56_21", unknown_segment = "0x00400f00", unknown_segbit = "27_1813" }
+{ unknown_bit = "00400f1b_56_24", unknown_segment = "0x00400f00", unknown_segbit = "27_1816" }
+{ unknown_bit = "00400f1b_45_19", unknown_segment = "0x00400f00", unknown_segbit = "27_1459" }
+{ unknown_bit = "00400f1b_45_20", unknown_segment = "0x00400f00", unknown_segbit = "27_1460" }
+{ unknown_bit = "00400f1b_45_21", unknown_segment = "0x00400f00", unknown_segbit = "27_1461" }
+{ unknown_bit = "00400f1b_45_24", unknown_segment = "0x00400f00", unknown_segbit = "27_1464" }
+{ unknown_bit = "00400f1b_66_19", unknown_segment = "0x00400f00", unknown_segbit = "27_2131" }
+{ unknown_bit = "00400f1b_66_20", unknown_segment = "0x00400f00", unknown_segbit = "27_2132" }
+{ unknown_bit = "00400f1b_66_21", unknown_segment = "0x00400f00", unknown_segbit = "27_2133" }
+{ unknown_bit = "00400f1b_86_19", unknown_segment = "0x00400f00", unknown_segbit = "27_2771" }
+{ unknown_bit = "00400f1b_86_20", unknown_segment = "0x00400f00", unknown_segbit = "27_2772" }
+{ unknown_bit = "00400f1b_66_24", unknown_segment = "0x00400f00", unknown_segbit = "27_2136" }
+{ unknown_bit = "00400f1b_86_21", unknown_segment = "0x00400f00", unknown_segbit = "27_2773" }
+{ unknown_bit = "00400f1b_86_24", unknown_segment = "0x00400f00", unknown_segbit = "27_2776" }
```

`prjxray-db` should be augmented with the missing bits, and the `minilitex` design should be used to investigate why these bits are present and not emitted from the `symbiflow` design.

mithro, 2020-02-12:
@litghost -- Should bit2fasm error with unknown bits these days?

litghost, 2020-02-12:
> @litghost -- Should bit2fasm error with unknown bits these days?

Valid question.  I think that might be a good default, with a way to disable the error.  For example, during prjxray fuzzing we just want the unknown bits printed, not an error.

litghost, 2020-10-29:
Duplicated by #1536

closed 2020-10-29 by litghost