f4pga/f4pga-arch-defs #1326: LV timing model workaround

litghost, 2020-02-20:
There is currently a workaround in place around the LV pip timing model.

There are two changes in prjxray for the workaround.  First is here:
https://github.com/SymbiFlow/prjxray/blob/84b1457b885861fee221c5c6255d41cfb962ffc0/utils/create_timing_worksheet_db.py#L147-L155

Second is here:
https://github.com/SymbiFlow/prjxray/blob/84b1457b885861fee221c5c6255d41cfb962ffc0/fuzzers/074-dump_all/jobtiles.tcl#L51-L64

And one change in symbiflow-arch-defs, here:
https://github.com/SymbiFlow/symbiflow-arch-defs/blob/2bf2ed6ca1f8f158fa8fa8a9c0d506f2ca284029/xc7/utils/prjxray_edge_library.py#L545-L551

When/if the LV pip timing model is improved, these workarounds will likely no longer be required.

litghost, 2020-02-20:
This issue will track any progress to improving the LV pip timing models, and work to remove the current hack-y solution that is present.  The current solution appears to generate error in the ~2-5% range, but further testing may show the workaround to behave worse in some circumstances.

mithro, 2020-02-20:
FYI - @daveshah1