f4pga/f4pga-arch-defs #1342: IOSETTINGS test failure

acomodi, 2020-02-26:
There are multiple CI runs that failed due to the `iosettings` tests. This happens on the VtR+SymbiFlow build and the failure is probably due to the router exhausting memory as it gets in an unroutable situation, having the target aborted as a result.

```
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Iter   Time    pres  BBs    Heap  Re-Rtd  Re-Rtd Overused RR Nodes      Wirelength      CPD       sTNS       sWNS       hTNS       hWNS Est Succ
      (sec)     fac Updt    push    Nets   Conns                                       (ns)       (ns)       (ns)       (ns)       (ns)     Iter
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Warning 63: No routing path for connection to sink_rr 1984856, retrying with full device bounding box
Warning 64: No routing path for connection to sink_rr 2022564, retrying with full device bounding box
   1    0.4     0.0    0 1811333     157     462     167 ( 0.005%)   53134 ( 0.7%)   14.348     -382.6    -14.348     -45.32     -5.348      N/A
   2    0.4     4.0    3 1714692      84     287      91 ( 0.003%)   53372 ( 0.7%)   16.271     -425.8    -16.271     -68.54     -7.150      N/A
   3    0.3     5.2    1 1085748      65     208      53 ( 0.001%)   53514 ( 0.7%)   16.661     -441.4    -16.661     -65.57     -7.432      N/A
   4    0.2     6.8    1  825292      46     142      31 ( 0.001%)   53621 ( 0.7%)   16.677     -432.1    -16.677     -73.34     -7.445      N/A
   5    0.3     8.8    1 1122030      31     100      22 ( 0.001%)   53547 ( 0.7%)   16.677     -437.0    -16.677     -69.53     -7.508      N/A
   6    0.3    11.4    1 1047153      19      56      12 ( 0.000%)   53725 ( 0.7%)   16.661     -437.4    -16.661     -69.03     -7.508      N/A
   7    0.3    14.9    1 1009861      11      24      10 ( 0.000%)   53587 ( 0.7%)   16.677     -438.7    -16.677     -67.91     -7.508      N/A
   8    0.3    19.3    1 1166944      10      26       5 ( 0.000%)   53717 ( 0.7%)   16.677     -437.8    -16.677     -68.75     -7.508      N/A
   9    0.3    25.1    1 1321944       6      16       6 ( 0.000%)   53606 ( 0.7%)   16.677     -437.1    -16.677     -69.49     -7.508      N/A
  10    0.4    32.6    1 1447286       5      16       3 ( 0.000%)   53776 ( 0.7%)   16.663     -436.5    -16.663     -70.21     -7.508       13
  11    0.4    42.4    1 1624490       3      11       2 ( 0.000%)   53790 ( 0.7%)   16.663     -437.3    -16.663     -69.47     -7.508       14
  12    0.7    55.1    1 2348395       3       7       2 ( 0.000%)   53775 ( 0.7%)   16.663     -437.7    -16.663     -69.07     -7.508       13
  13    0.4    71.7    1 1346309       3       6       2 ( 0.000%)   53628 ( 0.7%)   16.663     -437.6    -16.663     -69.09     -7.508       14
  14    0.9    93.2    1 2543806       3       6       2 ( 0.000%)   53775 ( 0.7%)   16.663     -438.4    -16.663     -68.34     -7.508       15
  15    0.4   121.1    1 1638553       3       6       2 ( 0.000%)   53628 ( 0.7%)   16.663     -437.6    -16.663     -69.09     -7.508       16
  16    0.8   157.5    1 2576775       3       6       2 ( 0.000%)   53775 ( 0.7%)   16.663     -438.4    -16.663     -68.34     -7.508       18
  17    0.5   204.7    1 1685493       3       6       2 ( 0.000%)   53628 ( 0.7%)   16.663     -437.6    -16.663     -69.09     -7.508       20
  18    0.4   266.2    1 1726536       3       6       4 ( 0.000%)   53777 ( 0.7%)   16.663     -438.0    -16.663     -68.77     -7.508       22
  19    1.3   346.0    1 4487077       4       9       3 ( 0.000%)   53624 ( 0.7%)   16.663     -438.3    -16.663     -68.39     -7.508      inf
  20    1.0   449.8    1 2738402       3       6       2 ( 0.000%)   53771 ( 0.7%)   16.663     -437.6    -16.663     -69.12     -7.508      inf
  21    0.6   584.8    1 2091772       4       7       3 ( 0.000%)   53633 ( 0.7%)   16.663     -437.6    -16.663     -69.13     -7.508      inf
  22    0.4   760.2    1 1435157       2       5       2 ( 0.000%)   53784 ( 0.7%)   16.663     -436.5    -16.663     -70.25     -7.508      inf
  23    1.4   988.3    1 4630075       4       8       2 ( 0.000%)   53631 ( 0.7%)   16.663     -438.7    -16.663     -68.00     -7.508      inf
  24    0.2  1284.7    1  389402       2       5       3 ( 0.000%)   53637 ( 0.7%)   16.663     -438.0    -16.663     -68.77     -7.508      inf
  25    1.3  1670.2    0 4347970       3       7       2 ( 0.000%)   53781 ( 0.7%)   16.663     -438.4    -16.663     -68.38     -7.508      inf
  26    0.6  2171.2    1 2178756       4       7       3 ( 0.000%)   53633 ( 0.7%)   16.663     -436.9    -16.663     -69.87     -7.508      inf
  27    0.4  2822.6    1 1561100       2       5       2 ( 0.000%)   53784 ( 0.7%)   16.663     -437.2    -16.663     -69.51     -7.508      inf
  28    0.9  3669.3    1 3439287       3       7       3 ( 0.000%)   53637 ( 0.7%)   16.663     -438.0    -16.663     -68.77     -7.508      inf
  29    1.5  4770.1    1 4887044       4       8       2 ( 0.000%)   53768 ( 0.7%)   16.663     -438.7    -16.663     -68.00     -7.508      inf
  30    0.9  6201.2    1 2686843       2       5       2 ( 0.000%)   53771 ( 0.7%)   16.663     -437.6    -16.663     -69.12     -7.508      inf
  31    0.6  8061.5    0 2274431       4       7       3 ( 0.000%)   53633 ( 0.7%)   16.663     -437.6    -16.663     -69.13     -7.508      132
  32    3.5 10480.0    1 1.7e+07       2       5       2 ( 0.000%)   53784 ( 0.7%)   16.663     -436.5    -16.663     -70.25     -7.508      555
  33    7.0 13624.0    1 3.5e+07       3       7       2 ( 0.000%)   53628 ( 0.7%)   16.663     -435.3    -16.663     -71.41     -7.508      106
  34    7.2 17711.2    1 2.6e+07       4       7       3 ( 0.000%)   53773 ( 0.7%)   16.663     -438.5    -16.663     -68.21     -7.508       93
  35    6.8 23024.5    1 2.7e+07       3       6       2 ( 0.000%)   53626 ( 0.7%)   16.663     -438.5    -16.663     -68.19     -7.508       98
  36    3.9 29931.8    1 1.9e+07       3       6       2 ( 0.000%)   53793 ( 0.7%)   16.663     -432.5    -16.663     -74.22     -7.508       89
  37    4.5 38911.4    1 2.2e+07       3       6       3 ( 0.000%)   53654 ( 0.7%)   16.663     -437.2    -16.663     -69.50     -7.508      135
  38    6.9 50584.8    1 3.4e+07       3       7       3 ( 0.000%)   53818 ( 0.7%)   16.663     -432.5    -16.663     -74.19     -7.508      332
  39   10.6 65760.3    1 4.5e+07       4       8       2 ( 0.000%)   53663 ( 0.7%)   16.663     -439.6    -16.663     -67.13     -7.508      inf
  40    3.7 85488.3    1 1.0e+07       2       5       2 ( 0.000%)   53664 ( 0.7%)   16.663     -436.1    -16.663     -70.61     -7.508      inf
  41    5.2 1.1e+05    0 2.3e+07       4       7       3 ( 0.000%)   53793 ( 0.7%)   16.663     -437.9    -16.663     -68.81     -7.508      311
  42    3.6 1.4e+05    1 1.8e+07       2       5       2 ( 0.000%)   53653 ( 0.7%)   16.663     -434.5    -16.663     -72.19     -7.508      inf
  43   10.9 1.9e+05    1 4.6e+07       4       8       2 ( 0.000%)   53792 ( 0.7%)   16.663     -438.4    -16.663     -68.28     -7.508      inf
  44    1.1 2.4e+05    1 4322095       2       5       3 ( 0.000%)   53793 ( 0.7%)   16.663     -432.1    -16.663     -74.66     -7.508     1868
  45    7.8 3.2e+05    0 3.9e+07       3       7       3 ( 0.000%)   53638 ( 0.7%)   16.663     -437.5    -16.663     -69.22     -7.508      inf
  46   10.2 4.1e+05    1 4.3e+07       4       8       2 ( 0.000%)   53775 ( 0.7%)   16.663     -433.4    -16.663     -73.37     -7.508      inf
  47    1.3 5.4e+05    1 5848738       1       4       2 ( 0.000%)   53777 ( 0.7%)   16.663     -435.0    -16.663     -71.75     -7.508     1120
  48    7.3 7.0e+05    0 3.6e+07       3       7       3 ( 0.000%)   53633 ( 0.7%)   16.663     -431.8    -16.663     -74.93     -7.508      385
  49   10.6 9.1e+05    1 4.6e+07       4       8       2 ( 0.000%)   53768 ( 0.7%)   16.663     -438.8    -16.663     -67.97     -7.508      inf
  50    1.2 1.2e+06    1 4820142       1       4       2 ( 0.000%)   53781 ( 0.7%)   16.663     -432.0    -16.663     -74.75     -7.508      inf
  51  243.9 1.5e+06    0 1.0e+09       4       8       3 ( 0.000%)   53668 ( 0.7%)   16.663     -438.1    -16.663     -68.64     -7.508      381
[... cut ...]
  58  291.3 9.6e+06    0 1.2e+09       3       7       2 ( 0.000%)   53655 ( 0.7%)   16.663     -435.7    -16.663     -71.01     -7.508      inf
  59   95.9 1.2e+07    0 5.0e+08       2       5       2 ( 0.000%)   53802 ( 0.7%)   16.663     -437.6    -16.663     -69.15     -7.508      inf
  60  222.0 1.6e+07    0 1.1e+09       4       8       3 ( 0.000%)   53659 ( 0.7%)   16.663     -432.6    -16.663     -74.17     -7.508      inf
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
ninja: build stopped: subcommand failed.
```

mkurc-ant, 2020-03-02:
@acomodi Does it happen every time? What's the SHA of the failing commit?

acomodi, 2020-03-02:
@mkurc-ant I wasn't able to reproduce this locally, saw this only in CI runs, didn't see this in a while though.

litghost, 2020-10-29:
@acomodi / @mkurc-ant Can we close this?

mkurc-ant, 2020-10-29:
I haven't encountered that error neither locally not in CI. Maybe it was due to some previous Kokoro configuration running too many parallel builds?

closed 2020-10-29 by litghost