f4pga/f4pga-arch-defs #1351: Complex design take long time to run through fasm2bels

acomodi, 2020-02-27:
There are complex designs such as the one introduced in https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1320, that take a huge amount of time to get placed and routed using Vivado.

I need to gather more data on what is the bottleneck of the process, but from local runs I can say the various bottlenecks are the following:

- parsing huge `.tcl` scripts in Vivado containing LOC and route constraints
- running DRC checks for each step (place, route, bitstream generation) in Vivado
- placing design in Vivado: even though everything is LOCed, Vivado placer still takes a huge amount of time to go through this step.

litghost, 2020-02-27:
So something I observed when debugging #1346 is that `write_bitstream` sometimes works without `place_design` and `route_design`.  This is likely a result of the input being fully constrained.  We could explore what it would take for a fully constrained design to always be provided to Vivado by commenting out the `place_design` and `route_design` options.

acomodi, 2020-02-27:
I will try to see if this solution works locally. If it is successful for the local runs, I can open a PR to remove place and route design from the runme.tcl

litghost, 2020-10-29:
This has been addressed both by improving the speed of Verilog/XDC and by adding interchange support.

closed 2020-10-29 by litghost