f4pga/f4pga-arch-defs #1352: Clock propagation story

litghost, 2020-02-27:
# Problem Statement

The currently way we define SDC files is fairly error prone and fragile.  When IOBUF inference was enabled, the SDC file for mini_ddr stopped worked.

The fundimental issue is that the SDC files we define use net names that get modified at multiple places in the flow.  There is also the need to propagate clocks through PLL's and MMCM's, which is less than convenient.

Some fixes should belong to the tools that modify nets, e.g. yosys and VPR.  But some of this needs to live outside of the tools, specifically propigation through PLL's and MMCM's.

# Proposed Solution

Create clock should only be **required** on input pins to the design.  Propagation through IBUF's, BUFG's, PLL's and MMCM's should be handled gracefully, generating additional constraints as needed.

1. User should specific the input clocks via `create_clock` (maybe in the XDC)?
2. Post-synthesis, the clocks should directly propagate through known clock buffers:
 - BUFR
 - BUFG
 - IBUF
 - etc
3. Clocks that pass through PLL's / MMCM's should be relationships established.
4. Output should be SDC suitable for input to VPR that matches blif.

acomodi, 2020-02-27:
So the basic idea is not to modify how VPR handles clocks, but rather automatically generate the SDC (maybe with a yosys plugin)?

litghost, 2020-02-27:
> So the basic idea is not to modify how VPR handles clocks, but rather automatically generate the SDC (maybe with a yosys plugin)?

I think both is needed (modifying VPR) and propigated.

litghost, 2020-10-29:
@tmichalak Once the SDC stuff is finished, please close this issue, thanks.

tmichalak, 2020-11-02:
Clocks are propagated through Yosys with this [SDC plugin PR](https://github.com/SymbiFlow/yosys-symbiflow-plugins/pull/27) and a [fix](https://github.com/SymbiFlow/yosys-symbiflow-plugins/pull/51).

closed 2020-11-02 by tmichalak