f4pga/f4pga-arch-defs #1359: Issue with clock net names in SDC

acomodi, 2020-03-03:
With the recent rework of I/OBUF, the net names relative to clocks nets (and also other I/OBUF corresponding nets).

By building https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1320 I get the following error:

```
Type: SDC file
File: /tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7/build/xc7/tests/soc/litex/base/arty.sdc
Line: 2
Message: Clock name or pattern 'clk100' does not correspond to any nets. To create a virtual clock, use the '-name' option.

Warning 60: Inferred implicit clock source $auto$iopadmap.cc:321:execute$79035.O[0] for netlist clock $abc$84833$auto$iopadmap.cc:361:execute$79077 (possibly data used as clock)
Warning 61: Timing edge from $auto$iopadmap.cc:321:execute$79035.I[0] to $auto$iopadmap.cc:321:execute$79035.O[0] will not be created since $auto$iopadmap.cc:321:execute$79035.O[0] has been identified as a clock generator
Warning 62: Inferred implicit clock source $auto$iopadmap.cc:321:execute$79037.O[0] for netlist clock eth_rx_clk (possibly data used as clock)
Warning 63: Timing edge from $auto$iopadmap.cc:321:execute$79037.I[0] to $auto$iopadmap.cc:321:execute$79037.O[0] will not be created since $auto$iopadmap.cc:321:execute$79037.O[0] has been identified as a clock generator
Warning 64: Inferred implicit clock source $auto$iopadmap.cc:321:execute$79038.O[0] for netlist clock eth_tx_clk (possibly data used as clock)
Warning 65: Timing edge from $auto$iopadmap.cc:321:execute$79038.I[0] to $auto$iopadmap.cc:321:execute$79038.O[0] will not be created since $auto$iopadmap.cc:321:execute$79038.O[0] has been identified as a clock generator
  Timing Graph Nodes: 119391
  Timing Graph Edges: 185415
  Timing Graph Levels: 42
# Build Timing Graph took 0.18 seconds (max_rss 25477.2 MiB, delta_rss +0.0 MiB)
Netlist contains 15 clocks
  Netlist Clock '$abc$84833$auto$iopadmap.cc:361:execute$79077' Fanout: 1 pins (0.0%), 1 blocks (0.0%)
  Netlist Clock 'eth_rx_clk' Fanout: 234 pins (0.2%), 234 blocks (0.9%)
  Netlist Clock 'eth_tx_clk' Fanout: 107 pins (0.1%), 107 blocks (0.4%)
  Netlist Clock 'soc_clk100bg' Fanout: 1 pins (0.0%), 1 blocks (0.0%)
  Netlist Clock 'sys_clk__VexRiscv.IBusCachedPlugin_cache.clk__VexRiscv.clk__VexRiscv.dataCache_1_.clk' Fanout: 6088 pins (5.1%), 6035 blocks (22.8%)
  Netlist Clock 'sys4x_clk' Fanout: 75 pins (0.1%), 59 blocks (0.2%)
  Netlist Clock 'sys4x_dqs_clk' Fanout: 2 pins (0.0%), 2 blocks (0.0%)
  Netlist Clock 'clk200_clk' Fanout: 8 pins (0.0%), 8 blocks (0.0%)
  Netlist Clock '$abc$84833$auto$iopadmap.cc:361:execute$79070' Fanout: 1 pins (0.0%), 1 blocks (0.0%)
  Netlist Clock 'soc_pll_fb' Fanout: 1 pins (0.0%), 1 blocks (0.0%)
  Netlist Clock 'soc_pll_sys' Fanout: 1 pins (0.0%), 1 blocks (0.0%)
  Netlist Clock 'soc_pll_sys4x' Fanout: 1 pins (0.0%), 1 blocks (0.0%)
  Netlist Clock 'soc_pll_sys4x_dqs' Fanout: 1 pins (0.0%), 1 blocks (0.0%)
  Netlist Clock 'soc_pll_clk200' Fanout: 1 pins (0.0%), 1 blocks (0.0%)
  Netlist Clock 'soc_pll_clk100' Fanout: 1 pins (0.0%), 1 blocks (0.0%)
# Load Timing Constraints
# Load Timing Constraints took 0.01 seconds (max_rss 25477.2 MiB, delta_rss +0.0 MiB)
The entire flow of VPR took 72.94 seconds (max_rss 25477.2 MiB)
ninja: build stopped: subcommand failed.

[ID: 5803756] Build finished after 3958 secs, exit value: 1
```

Where `$abc$84833$auto$iopadmap.cc:361:execute$79077` should be relative to the `clk100` clock net.

mkurc-ant, 2020-03-03:
I looked into it trying to find a way to make Yosys somehow propagate the top-level net name through an IBUF and use it to format name of the net that goes out of it. But I couldn't find a solution.

We could make a plugin for Yosys that does just that. Then we could have eg. for a clock input named `clk_100`  the name of the actual net that has to be constrained in a form like `$clk_100_ibuf`.

Another solution would be to teach VPR how to propagate the clock constraint through an IBUF. But that would have to be done in a generic way.

litghost, 2020-03-03:
This issue is already described in #1352

closed 2020-03-03 by litghost