f4pga/f4pga-arch-defs #1370: make env failing

jchidley, 2020-03-12:
Today, I followed the [Getting Started Instructions](https://symbiflow.readthedocs.io/en/latest/symbiflow-arch-defs/docs/source/getting-started.html)

the step `make env` failed.

Here is the output of "C:/Users/jackc/git/symbiflow-arch-defs/build/CMakeFiles/CMakeError.log"

```cmd
Determining if the include file pthread.h exists failed with the following output:
Change Dir: C:/Users/jackc/git/symbiflow-arch-defs/build/CMakeFiles/CMakeTmp

Run Build Command(s):C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/MSBuild/Current/Bin/MSBuild.exe cmTC_7c8f0.vcxproj /p:Configuration=Debug /p:Platform=x64 /p:VisualStudioVersion=16.0 /v:m
Microsoft (R) Build Engine version 16.4.0+e901037fe for .NET Framework

Copyright (C) Microsoft Corporation. All rights reserved.

  Microsoft (R) C/C++ Optimizing Compiler Version 19.24.28319 for x64

  Copyright (C) Microsoft Corporation.  All rights reserved.

  CheckIncludeFile.c

  cl /c /Zi /W3 /WX- /diagnostics:column /Od /Ob0 /D WIN32 /D _WINDOWS /D "CMAKE_INTDIR=\"Debug\"" /D _MBCS /Gm- /RTC1 /MDd /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /Fo"cmTC_7c8f0.dir\Debug\\" /Fd"cmTC_7c8f0.dir\Debug\vc142.pdb" /Gd /TC /errorReport:queue "C:\Users\jackc\git\symbiflow-arch-defs\build\CMakeFiles\CMakeTmp\CheckIncludeFile.c"

C:\Users\jackc\git\symbiflow-arch-defs\build\CMakeFiles\CMakeTmp\CheckIncludeFile.c(1,10): fatal error C1083: Cannot open include file: 'pthread.h': No such file or directory [C:\Users\jackc\git\symbiflow-arch-defs\build\CMakeFiles\CMakeTmp\cmTC_7c8f0.vcxproj]

Performing C++ SOURCE FILE Test CXX_COMPILER_SUPPORTS_UNDEF_VAR failed with the following output:
Change Dir: C:/Users/jackc/git/symbiflow-arch-defs/build/CMakeFiles/CMakeTmp

Run Build Command(s):C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/MSBuild/Current/Bin/MSBuild.exe cmTC_a8066.vcxproj /p:Configuration=Debug /p:Platform=x64 /p:VisualStudioVersion=16.0 /v:m
Microsoft (R) Build Engine version 16.4.0+e901037fe for .NET Framework

Copyright (C) Microsoft Corporation. All rights reserved.

  Microsoft (R) C/C++ Optimizing Compiler Version 19.24.28319 for x64

  Copyright (C) Microsoft Corporation.  All rights reserved.

  cl /c /Zi /W3 /WX- /diagnostics:column /Od /Ob0 /D WIN32 /D _WINDOWS /D CXX_COMPILER_SUPPORTS_UNDEF_VAR /D "CMAKE_INTDIR=\"Debug\"" /D _MBCS /Gm- /EHsc /RTC1 /MDd /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /GR /Fo"cmTC_a8066.dir\Debug\\" /Fd"cmTC_a8066.dir\Debug\vc142.pdb" /Gd /TP /errorReport:queue  -Wundefined-var-template "C:\Users\jackc\git\symbiflow-arch-defs\build\CMakeFiles\CMakeTmp\src.cxx"

cl : command line error D8021: invalid numeric argument '/Wundefined-var-template' [C:\Users\jackc\git\symbiflow-arch-defs\build\CMakeFiles\CMakeTmp\cmTC_a8066.vcxproj]

Source file was:
int main() { return 0; }
```

I am not a C or C++ developer.  So these errors mean nothing to me.

acomodi, 2020-03-12:
Hi @jchidley, thanks for filing this.

I suppose the OS you are using is Windows, correct?
In that case, do you have access to a Linux machine?

jchidley, 2020-03-12:
The installation on Arch Linux failed for a different reason (after I had fixed the lack of `cmake`)

acomodi, 2020-03-12:
@jchidley Ok, can you please switch back to Arch Linux and print the error log you get there?

jchidley, 2020-03-12:
@acomodi Arch fails on the `make all_conda` step.  Do you still want those logs as they appear to have nothing to do with this error?

acomodi, 2020-03-12:
@jchidley Sure, logs always contain helpful information to understand what is going wrong. If the log is too long you may zip it and drop it here.

jchidley, 2020-03-12:
@acomodi I am confused.  I have a completely separate Arch Linux machine that is having a different problem that I would file a different issue for.  I want to install the tools on my Windows machine.

acomodi, 2020-03-12:
@jchidley The problem is that, for the time being, there is no Windows support. `make all_conda` downloads Linux binaries that are incompatible in Windows.

It is possible that that Arch Linux has different core libraries versions that may result in the error you mentioned, but it would be better to check what the error-log is in Arch Linux.

Best would be to use Debian-based distros.

jchidley, 2020-03-12:
@acomodi In that case, I suggest that you add "Only works on ubuntu linux" as part of your *Getting Started* instructions.

0anubis0, 2020-03-21:
I had problem with the make env as well.
I have Ubuntu 18.04.4 LTS
First of all, I had to update the Cmake from 3.10.1 to 3.12.4 with manual compile.
I got the following error after make env: (pthread not found)

```
Determining if the pthread_create exist failed with the following output:
Change Dir: /home/rndc/symbiflow-arch-defs/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_e54a2/fast"
make[1]: Entering directory '/data/home/rndc/symbiflow-arch-defs/build/CMakeFiles/CMakeTmp'
/usr/bin/make -f CMakeFiles/cmTC_e54a2.dir/build.make CMakeFiles/cmTC_e54a2.dir/build
make[2]: Entering directory '/data/home/rndc/symbiflow-arch-defs/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_e54a2.dir/CheckSymbolExists.c.o
/usr/bin/cc    -o CMakeFiles/cmTC_e54a2.dir/CheckSymbolExists.c.o   -c /home/rndc/symbiflow-arch-defs/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTC_e54a2
/usr/local/bin/cmake -E cmake_link_script CMakeFiles/cmTC_e54a2.dir/link.txt --verbose=1
/usr/bin/cc      -rdynamic CMakeFiles/cmTC_e54a2.dir/CheckSymbolExists.c.o  -o cmTC_e54a2 
CMakeFiles/cmTC_e54a2.dir/CheckSymbolExists.c.o: In function `main':
CheckSymbolExists.c:(.text+0x1b): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_e54a2.dir/build.make:86: recipe for target 'cmTC_e54a2' failed
make[2]: *** [cmTC_e54a2] Error 1
make[2]: Leaving directory '/data/home/rndc/symbiflow-arch-defs/build/CMakeFiles/CMakeTmp'
Makefile:121: recipe for target 'cmTC_e54a2/fast' failed
make[1]: *** [cmTC_e54a2/fast] Error 2
make[1]: Leaving directory '/data/home/rndc/symbiflow-arch-defs/build/CMakeFiles/CMakeTmp'

File /home/rndc/symbiflow-arch-defs/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/rndc/symbiflow-arch-defs/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_0081a/fast"
make[1]: Entering directory '/data/home/rndc/symbiflow-arch-defs/build/CMakeFiles/CMakeTmp'
/usr/bin/make -f CMakeFiles/cmTC_0081a.dir/build.make CMakeFiles/cmTC_0081a.dir/build
make[2]: Entering directory '/data/home/rndc/symbiflow-arch-defs/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_0081a.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_0081a.dir/CheckFunctionExists.c.o   -c /usr/local/share/cmake-3.12/Modules/CheckFunctionExists.c
Linking C executable cmTC_0081a
/usr/local/bin/cmake -E cmake_link_script CMakeFiles/cmTC_0081a.dir/link.txt --verbose=1
/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create    -rdynamic CMakeFiles/cmTC_0081a.dir/CheckFunctionExists.c.o  -o cmTC_0081a -lpthreads 
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_0081a.dir/build.make:86: recipe for target 'cmTC_0081a' failed
make[2]: *** [cmTC_0081a] Error 1
make[2]: Leaving directory '/data/home/rndc/symbiflow-arch-defs/build/CMakeFiles/CMakeTmp'
Makefile:121: recipe for target 'cmTC_0081a/fast' failed
make[1]: *** [cmTC_0081a/fast] Error 2
make[1]: Leaving directory '/data/home/rndc/symbiflow-arch-defs/build/CMakeFiles/CMakeTmp'

Performing C++ SOURCE FILE Test CXX_COMPILER_SUPPORTS_UNDEF_VAR failed with the following output:
Change Dir: /home/rndc/symbiflow-arch-defs/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_483a5/fast"
make[1]: Entering directory '/data/home/rndc/symbiflow-arch-defs/build/CMakeFiles/CMakeTmp'
/usr/bin/make -f CMakeFiles/cmTC_483a5.dir/build.make CMakeFiles/cmTC_483a5.dir/build
make[2]: Entering directory '/data/home/rndc/symbiflow-arch-defs/build/CMakeFiles/CMakeTmp'
Building CXX object CMakeFiles/cmTC_483a5.dir/src.cxx.o
/usr/bin/c++    -DCXX_COMPILER_SUPPORTS_UNDEF_VAR   -Wundefined-var-template -o CMakeFiles/cmTC_483a5.dir/src.cxx.o -c /home/rndc/symbiflow-arch-defs/build/CMakeFiles/CMakeTmp/src.cxx
c++: error: unrecognized command line option '-Wundefined-var-template'; did you mean '-Wundefined-do-loop'?
CMakeFiles/cmTC_483a5.dir/build.make:65: recipe for target 'CMakeFiles/cmTC_483a5.dir/src.cxx.o' failed
make[2]: *** [CMakeFiles/cmTC_483a5.dir/src.cxx.o] Error 1
make[2]: Leaving directory '/data/home/rndc/symbiflow-arch-defs/build/CMakeFiles/CMakeTmp'
Makefile:121: recipe for target 'cmTC_483a5/fast' failed
make[1]: *** [cmTC_483a5/fast] Error 2
make[1]: Leaving directory '/data/home/rndc/symbiflow-arch-defs/build/CMakeFiles/CMakeTmp'

Source file was:
int main() { return 0; }
```

0anubis0, 2020-03-21:
Using older commit worked:
Commits on Mar 11, 2020 : 318b4806c5982b525c6ecb4387578212d482d97d

acomodi, 2020-03-23:
@0anubis0 Thanks for reporting this.
I had a look at the issue and I think I found the reasons behind this error.

The problem is in the thread libraries that CMake is searching for in the system.
The `pthread_create` function is (at least in Ubuntu) found in the `pthread` library. The problem is that CMake is searching it also in other libraries, such as `pthreads`, which does not have that.

I switched my CMake version to 3.12.4 as well and got the same error as you get.
Switching back to CMake 3.14.0 (which I previously had) did let me pass this Error.

What I get though is still:
```
-- No CMAKE_CXX_STANDARD set, assuming 11
-- Performing Test CXX_COMPILER_SUPPORTS_UNDEF_VAR
-- Performing Test CXX_COMPILER_SUPPORTS_UNDEF_VAR - Failed
```

This does not produce a fatal error though, because I presume that the CMake build eventually finds the `pthread_create` function in the correct pthread library. Only thing with the `3.12.4` CMake version could be that, if the function is not found, the CMake build just fails.

I suggest, for now, to switch to a later CMake version and test whether it passes the `make env` step.
I will open a PR to change the minimum CMake version.

acomodi, 2020-03-28:
I believe this has been solved with https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1379. Closing

closed 2020-03-28 by acomodi