f4pga/f4pga-arch-defs #1375: Blink example not worked

0anubis0, 2020-03-22:
318b4806c5982b525c6ecb4387578212d482d97d
The blink example for icestick not worked. Icestorm was not found.

I have installed the icestorm, after I was able to program the icestick but the result was bullshit. Just one led with half power.
-Tested the iceprog with a lattice example, worked.
-Modified the example, static declarations worked
-Everything related to clock not worked.

One note, at the build directory there was a synthetised verilog file. Looks like the yosys was optimized out the processing part.

The example worked out of box with the following commands:

```
yosys -p "synth_ice40 -blif rot.blif" rot.v
arachne-pnr -d 1k -p rot.pcf rot.blif -o rot.asc
icepack rot.asc rot.bin
iceprog rot.bin
```

litghost, 2020-03-22:
The ice40 portion of symbiflow-arch-defs has had very little development, with the majority of the focus on Xilinx 7-series support.  This is not super suprising given where the focus has been.

litghost, 2020-10-29:
Won't fix

closed 2020-10-29 by litghost