f4pga/f4pga-arch-defs #1424: Branch : Quicklogic : Error during generation of top_bit.v file

rakeshm75, 2020-04-16:
We get the following error when running design3-ql-chandalar_bit_v command: 

Generating design3-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top_bit.v
Traceback (most recent call last):
  File "/home/rakeshm/symbiflow-arch-defs/quicklogic/utils/fasm2bels.py", line 554, in <module>
    verilog, pcf, qcf = f2b.convert_to_verilog(fasmlines)
  File "/home/rakeshm/symbiflow-arch-defs/quicklogic/utils/fasm2bels.py", line 487, in convert_to_verilog
    self.resolve_connections()
  File "/home/rakeshm/symbiflow-arch-defs/quicklogic/utils/fasm2bels.py", line 406, in resolve_connections
    self.resolve_hops()
  File "/home/rakeshm/symbiflow-arch-defs/quicklogic/utils/fasm2bels.py", line 388, in resolve_hops
    hop = get_name_and_hop(self.designhops[tloc][hop[0]])
KeyError: 'WBs_ADR[2]'
make[3]: *** [quicklogic/tests/quicklogic_testsuite/design3/design3-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top_bit.v] Error 1
make[2]: *** [quicklogic/tests/quicklogic_testsuite/design3/CMakeFiles/file_quicklogic_tests_quicklogic_testsuite_design3_design3-ql-chandalar_ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp_top_bit.v.dir/all] Error 2
make[1]: *** [quicklogic/tests/quicklogic_testsuite/design3/CMakeFiles/design3-ql-chandalar_bit_v.dir/rule] Error 2
make: *** [quicklogic/tests/quicklogic_testsuite/design3/CMakeFiles/design3-ql-chandalar_bit_v.dir/rule] Error 2

[design3.zip](https://github.com/SymbiFlow/symbiflow-arch-defs/files/4486781/design3.zip)

kgugala, 2020-04-21:
I cannot really reproduce this one. The design is optimized out and I get very simple top_bit.v file. I'll check why the design is optimized out

rakeshm75, 2020-04-21:
@kgugala Are you using the attached design (design3.zip) files. I have pulled debug signals (in the attached design) out, now I am able to place and route the design, tool is not optimizing the design. I have attached the design along with eblif, JLINK files. But when I run design3-ql-chandalar_bit_v command I get the above error.

[design3_impl.zip](https://github.com/SymbiFlow/symbiflow-arch-defs/files/4510269/design3_impl.zip)

kgugala, 2020-04-22:
thanks

I also investigated the design and figured out what was causing the problem here. The original code have `pragma attribute         preserve_cell true` set on some blocks. This is, I assume, disabling optimization on those in the tool you're using. 

Yosys does not support those non-standard pragmas, but rather use `(* keep *)` attributes. I added the attributes for the modules in this commit https://github.com/antmicro/symbiflow-arch-defs/commit/f2777337754251b0d16f6e273a5eb6f39ed61135. This prevents the design to be optimized out.

> But when I run design3-ql-chandalar_bit_v command I get the above error.

In the latest fasm2bels code this is fixed. Please fetch the newest version

rakeshm75, 2020-04-22:
@kgugala Thanks!
This issue is resolved, so closing the issue.

closed 2020-04-22 by rakeshm75