f4pga/f4pga-arch-defs #1429: Vendor Tools IO placement error

acomodi, 2020-04-18:
The following error shows up from time to time on the Vendor tools CI, regarding the baselitex test:

```
FAILED: cd /tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7-vendor/build/xc/xc7/tests/soc/litex/base/baselitex_arty/artix7-xc7a50t-virt-xc7a50t-test && /usr/bin/cmake -E remove -f baselitex_arty/artix7-xc7a50t-virt-xc7a50t-test/design_baselitex_arty_vivado.dcp && /usr/bin/cmake -E remove -f baselitex_arty/artix7-xc7a50t-virt-xc7a50t-test/design_baselitex_arty_vivado.xpr && /tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7-vendor/third_party/prjxray/utils/vivado.sh -mode batch -source /tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7-vendor/build/xc/xc7/tests/soc/litex/base/baselitex_arty_vivado_runme.tcl > /tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7-vendor/build/xc/xc7/tests/soc/litex/base/baselitex_arty/artix7-xc7a50t-virt-xc7a50t-test/vivado.stdout.log
ERROR: [Place 30-294] The following IO terminals are loads of BUFIO/BUFR/BUFH
 and they should be placed to the same clock region as BUFIO/BUFR/BUFH instance.
 They have incompatible IO Standards that makes their placement illegal
 IO Standard1: SioStd: DIFF_SSTL135 VCCO = 1.35 Termination: 0  TermDir:  Out  BufioId: 2 Bank: 34 Placed LVDS 
     List of IO terminals :
	Term: ddram_dqs_p[0]
	Term: ddram_dqs_n[0]
	Term: ddram_dqs_p[1]
	Term: ddram_dqs_n[1]
	Term: ddram_clk_p
	Term: ddram_clk_n

 IO Standard2: SioStd: LVCMOS33   VCCO = 3.3 Termination: 0  TermDir:  Out  BufioId: 2 Bank: 15 Drv: 12 Placed 
     List of IO terminals :
	Term: eth_ref_clk

ERROR: [Place 30-374] IO placer failed to find a solution
Below is the partial placement that can be analyzed to see if any constraint modifications will make the IO placement problem easier to solve.

+-------------------------------------------------------------------------------------------------------------------------------------------------------------------------+
|                                                                     IO Placement : Bank Stats                                                                           |
+----+-------+-------+------------------------------------------------------------------------+------------------------------------------+--------+--------+--------+-----+
| Id | Pins  | Terms |                               Standards                                |                IDelayCtrls               |  VREF  |  VCCO  |   VR   | DCI |
+----+-------+-------+------------------------------------------------------------------------+------------------------------------------+--------+--------+--------+-----+
|  0 |     0 |     0 |                                                                        |                                          |        |        |        |     |
| 14 |    50 |     2 | LVCMOS33(2)                                                            |                                          |        |  +3.30 |    YES |     |
| 15 |    50 |    16 | LVCMOS33(16)                                                           |                                          |        |  +3.30 |    YES |     |
| 16 |    10 |     2 | LVCMOS33(2)                                                            |                                          |        |  +3.30 |    YES |     |
| 34 |    50 |    48 | SSTL135(42)  DIFF_SSTL135(6)                                           | _IOI3_X113Y26_IDELAYCTRL_X1Y0_IDELAYCTRL |  +0.68 |  +1.35 |    YES |     |
| 35 |    50 |     4 | LVCMOS33(4)                                                            |                                          |        |  +3.30 |    YES |     |
+----+-------+-------+------------------------------------------------------------------------+------------------------------------------+--------+--------+--------+-----+
|    |   210 |    72 |                                                                        |                                          |        |        |        |     |
+----+-------+-------+------------------------------------------------------------------------+------------------------------------------+--------+--------+--------+-----+

IO Placement:
+--------+----------------------+-----------------+----------------------+----------------------+----------------------+
| BankId |             Terminal | Standard        | Site                 | Pin                  | Attributes           |
+--------+----------------------+-----------------+----------------------+----------------------+----------------------+
| 14     | led[2]               | LVCMOS33        | IOB_X0Y2             | T9                   |                      |
|        | led[3]               | LVCMOS33        | IOB_X0Y1             | T10                  |                      |
+--------+----------------------+-----------------+----------------------+----------------------+----------------------+
| 15     | eth_clocks_rx        | LVCMOS33        | IOB_X0Y72            | F15                  |                      |
|        | eth_clocks_tx        | LVCMOS33        | IOB_X0Y74            | H16                  |                      |
|        | eth_mdc              | LVCMOS33        | IOB_X0Y71            | F16                  |                      |
|        | eth_mdio             | LVCMOS33        | IOB_X0Y66            | K13                  |                      |
|        | eth_ref_clk          | LVCMOS33        | IOB_X0Y56            | G18                  |                      |
|        | eth_rst_n            | LVCMOS33        | IOB_X0Y60            | C16                  |                      |
|        | eth_rx_data[0]       | LVCMOS33        | IOB_X0Y57            | D18                  |                      |
|        | eth_rx_data[1]       | LVCMOS33        | IOB_X0Y68            | E17                  |                      |
|        | eth_rx_data[2]       | LVCMOS33        | IOB_X0Y58            | E18                  |                      |
|        | eth_rx_data[3]       | LVCMOS33        | IOB_X0Y63            | G17                  |                      |
|        | eth_rx_dv            | LVCMOS33        | IOB_X0Y73            | G16                  |                      |
|        | eth_tx_data[0]       | LVCMOS33        | IOB_X0Y70            | H14                  |                      |
|        | eth_tx_data[1]       | LVCMOS33        | IOB_X0Y62            | J14                  |                      |
|        | eth_tx_data[2]       | LVCMOS33        | IOB_X0Y65            | J13                  |                      |
|        | eth_tx_data[3]       | LVCMOS33        | IOB_X0Y64            | H17                  |                      |
|        | eth_tx_en            | LVCMOS33        | IOB_X0Y61            | H15                  | *                    |
+--------+----------------------+-----------------+----------------------+----------------------+----------------------+
| 16     | serial_rx            | LVCMOS33        | IOB_X0Y121           | A9                   |                      |
|        | serial_tx            | LVCMOS33        | IOB_X0Y111           | D10                  | *                    |
+--------+----------------------+-----------------+----------------------+----------------------+----------------------+
| 34     | ddram_clk_p          | DIFF_SSTL135    | IOB_X1Y8             | U9                   |                      |
|        | ddram_clk_n          | DIFF_SSTL135    | IOB_X1Y7             | V9                   |                      |
|        | ddram_dqs_p[0]       | DIFF_SSTL135    | IOB_X1Y44            | N2                   |                      |
|        | ddram_dqs_n[0]       | DIFF_SSTL135    | IOB_X1Y43            | N1                   |                      |
|        | ddram_dqs_p[1]       | DIFF_SSTL135    | IOB_X1Y32            | U2                   |                      |
[... 28 lines cut ...]
|        | ddram_dq[14]         | SSTL135         | IOB_X1Y33            | U3                   | VRef=+0.68           |
|        | ddram_dq[15]         | SSTL135         | IOB_X1Y28            | R3                   | VRef=+0.68           |
|        | ddram_dq[1]          | SSTL135         | IOB_X1Y45            | L3                   | VRef=+0.68           |
|        | ddram_dq[2]          | SSTL135         | IOB_X1Y46            | K3                   | VRef=+0.68           |
[... cut ...]
|        | cpu_reset            | LVCMOS33        | IOB_X1Y68            | C2                   |                      |
|        | led[0]               | LVCMOS33        | IOB_X1Y51            | H5                   |                      |
|        | led[1]               | LVCMOS33        | IOB_X1Y50            | J5                   |                      |
+--------+----------------------+-----------------+----------------------+----------------------+----------------------+

```

Still unknown the reason behind this behaviour which seems to be "randomic"

acomodi, 2020-04-28:
To reproduce the issue the following bitstream can be used in the fasm2bels step:
[top.bit.zip](https://github.com/SymbiFlow/symbiflow-arch-defs/files/4547299/top.bit.zip)

litghost, 2020-07-06:
```
ERROR: [Place 30-294] The following IO terminals are loads of BUFIO/BUFR/BUFH
 and they should be placed to the same clock region as BUFIO/BUFR/BUFH instance.
 They have incompatible IO Standards that makes their placement illegal
 IO Standard1: SioStd: DIFF_SSTL135 VCCO = 1.35 Termination: 0  TermDir:  Out  BufioId: 2 Bank: 34 Placed LVDS
```

This error appears to be something that should be fixed in the clock placement logic.