f4pga/f4pga-arch-defs #1431: Branch : Quicklogic : Different ports and port names in top_bit.v file compared to the design file

rakeshm75, 2020-04-20:
Different ports and port names in top_bit.v file compared to the rtl. The ports in the top_bit.v file of the design bin2seven:

    inout X10Y3_inout,
    inout X12Y32_inout,
    inout X12Y3_inout,
    inout X14Y32_inout,
    inout X14Y3_inout,
    inout X16Y32_inout,
    inout X18Y32_inout,
    inout X20Y32_inout,
    inout X22Y32_inout,
    inout X24Y32_inout,
    inout X4Y3_inout,
    inout X6Y3_inout,
    inout X8Y3_inout

In the rtl the ports are as following: 

    input clock_100Mhz, 
    input reset, 
    output reg [3:0] Anode_Activate, 
[bit2seven_design.zip](https://github.com/SymbiFlow/symbiflow-arch-defs/files/4502642/bit2seven_design.zip)

    output reg [6:0] LED_out

kgugala, 2020-04-22:
this feature has been added in https://github.com/antmicro/symbiflow-arch-defs/commit/042922d3d2cabbbeb17f5a3a373b3d74f04bbe0e

Please fetch the code and rerun the test

rakeshm75, 2020-04-22:
@kgugala  I did pull the new code and run the test (with different design), I get the following error:

Traceback (most recent call last):
  File "/adhome/rakeshm/symbiflow-arch-defs/quicklogic/utils/fasm2bels.py", line 569, in <module>
    pcf_data = parse_pcf(args.input_pcf)
NameError: name 'parse_pcf' is not defined
make[3]: *** [quicklogic/tests/quicklogic_testsuite/counter_32bit/counter_32bit-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top_bit.v] Error 1
make[2]: *** [quicklogic/tests/quicklogic_testsuite/counter_32bit/CMakeFiles/file_quicklogic_tests_quicklogic_testsuite_counter_32bit_counter_32bit-ql-chandalar_ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp_top_bit.v.dir/all] Error 2
make[1]: *** [quicklogic/tests/quicklogic_testsuite/counter_32bit/CMakeFiles/counter_32bit-ql-chandalar_bit_v.dir/rule] Error 2
make: *** [quicklogic/tests/quicklogic_testsuite/counter_32bit/CMakeFiles/counter_32bit-ql-chandalar_bit_v.dir/rule] Error 2

rakeshm75, 2020-04-22:
I get the same error with bin2seven design as well:
  File "/adhome/rakeshm/symbiflow-arch-defs/quicklogic/utils/fasm2bels.py", line 569, in <module>
    pcf_data = parse_pcf(args.input_pcf)
NameError: name 'parse_pcf' is not defined
make[3]: *** [quicklogic/tests/quicklogic_testsuite/bin2seven/bin2seven-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top_bit.v] Error 1
make[2]: *** [quicklogic/tests/quicklogic_testsuite/bin2seven/CMakeFiles/file_quicklogic_tests_quicklogic_testsuite_bin2seven_bin2seven-ql-chandalar_ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp_top_bit.v.dir/all] Error 2
make[1]: *** [quicklogic/tests/quicklogic_testsuite/bin2seven/CMakeFiles/bin2seven-ql-chandalar_bit_v.dir/rule] Error 2
make: *** [quicklogic/tests/quicklogic_testsuite/bin2seven/CMakeFiles/bin2seven-ql-chandalar_bit_v.dir/rule] Error 2

rakeshm75, 2020-04-23:
@kgugala, Yes I don't get the above errors now.

But when the IO ports are generated from PCF file, there are issues:
1. Vector (bus signals) signals are represented as individual signals
            Ex: output [3:0] count is represented as
                  inout count(0)
                  inout count(1)
                  inout count(2)
                  inout count(3)
2. Simulator will throw syntax error with () brackets
        module top (
                                 inout clk,
                                 inout comb_out,
                                 inout count(0),
                                 inout count(1),
                                 inout count(2),
                                 inout count(3),
                                 inout enable,
                                 inout set);

In SpDE tool flow, the synthesis tool generates a .vh file from the top level rtl file. This .vh file contains only the port definition. This used for the .vq generation (post layout netlist).

module top ( count, clk, enable, set, comb_out ) ;
    output [3:0]count ;
    input clk ;
    input enable ;
    input set ;
    output comb_out ;

kgugala, 2020-04-23:
this is now fixed

tpagarani, 2020-04-23:
Hi @kgugala, Does the fix take care of generating the vector signal as well or only syntax fix for the bracket? while this fix will resolve this specific issue, as discussed please help come up with solution where there is no dependency on pcf file

kgugala, 2020-05-06:
yes, vectors are generated correctly

closed 2020-05-06 by kgugala

kgugala, 2020-05-06:
uni-directional ports are now generated without buffers