f4pga/f4pga-arch-defs #1436: Branch : Quicklogic : Primitives for the post-layout simulations

rakeshm75, 2020-04-22:
Which are the primitive files that can be used for post-layout simulations?

Are the files @ /symbiflow-arch-defs/quicklogic/primitives used for post-layout simulations?

If yes, the primitive names in the top_bit.v file does not match with the names in the above files.
   Ex. logic_cell_macro in top_bit.v file and LOGIC in the above primitive file

If the above primitive files are not used for post-layout simulations then can you please point me to the relevant primitive files used for post-layout simulations.

rakeshm75, 2020-04-23:
There is port miss match in the primitives that are in top_bit.v file and the above mentioned primitive files (@ /symbiflow-arch-defs/quicklogic/primitives).

Can you point me to the primitives files that needs to be used along with top_bit.v (post layout netlist) for running post-layout simulations.

kgugala, 2020-04-23:
hi @rakeshm75 

please use the `cells_sim` library from Yosys https://github.com/antmicro/yosys/blob/quicklogic-rebased/techlibs/quicklogic/cells_sim.v

tpagarani, 2020-04-23:
hi @kgugala ,

primitives in cells_sim.v don't have any provision for timing annotation whereas /symbiflow-arch-defs/quicklogic/primitives has timing constructs emdedded. 

Is that work in progress? let us know when would that be available as well as SDF generation.