f4pga/f4pga-arch-defs #1437: quicklogic: investigate negative cost warnings

kgugala, 2020-04-22:
During routing VPR prints a lot of negative cost warnings:
```
Warning 14404762: Cost is negative? cost = -2.352, bucket = -2147483648
```

It is unclear if this is caused by the timing data we use, bug in timing import or a bug in timing model.

tpagarani, 2020-04-22:
We see that route.log is taking up to 2GB space with these warning messages. 
Could this also be causing increased run time in router?

litghost, 2020-04-22:
The cost function can never be negative in a sensical timing model.  A negative cost is a signal that travels backwards in time.

litghost, 2020-04-22:
Also the units on the timing driven routers is seconds, so a cost of -2.3 seconds is pretty crazy.

kgugala, 2020-05-06:
This has been fixed by bumping VPR version. Negative costs do not appear in the routing anymore.

closed 2020-05-06 by kgugala