f4pga/f4pga-arch-defs #1441: QuickLogic: incomplete generated sdf files

lnsharma, 2020-04-23:
While going through the generated sdf files present at location: symbiflow-arch-defs/build/quicklogic/devices/ql-eos-s3-virt/sdf, I noticed the following issues:

1. SDFs has delays only for worst corner. Best corner is not yet considered though inputs timing files are present at: "symbiflow-arch-defs/third_party/ql-eos-s3/Timing Data Files"
2. Generated SDF file for routing delay "QLAL4S3B_ss_0p990v_m040c.sdf" is empty
3. In sdf file - "QMUX_ss_0p990v_m040c.sdf", I noticed the following:
     - sdf condition is mentioned as "CELLTYPE", wondering if this could lead to various instances in post-layout file
    - Delays are corresponding to worst case, so aren't they be mentioned as (0.17: : ) rather than ( :0.17: )
    - This sdf file doesn't consider timing sense, so wondering how it will account for rr, ff, rf & fr delays
    - How will you differentiate between positive_unate and negative_unate delay data in these asdfs
    - These points also apply to other sdf files related to clock cells

kgugala, 2020-04-29:
hi @lnsharma as for the 2. this PR https://github.com/antmicro/quicklogic-timings-importer/pull/1 should add functionality to import the timing data from liberty format used in `QLAL4S3B_ss_0p990v_m040c.sdf`. I'll proceed with merge of that PR and bump the submodule.

kgugala, 2020-05-06:
@lnsharma the timing importer script has been updated and it can handle the liberty library files now. The `QLAL4S3B_ss_0p990v_m040c.sdf` is now imported.

Since the importer has been updated and many issues are now fixed, please grab the latest code , regenerate the sdfs and check them again.

lnsharma, 2020-05-06:
Any comments on item #3. Also how should we report both best and worst case delays as mentioned in item #1.

lnsharma, 2020-05-07:
I was reviewing generated sdf file - QLAL4S3B_ss_0p990v_m040c.sdf and as mentioned in my initial comment #3.i about sdf condition, I think we should define the condition is sdf defined format. 
For example, sdf file contains these lines:

(CELL
        (CELLTYPE "GMUX_IS0_EQ_0")
        (INSTANCE GMUX)
        (DELAY
            (ABSOLUTE
                (IOPATH IP IZ (:0.445:)(:0.432:))
            )
        )
    )
    (CELL
        (CELLTYPE "GMUX_IS0_EQ_1")
        (INSTANCE GMUX)
        (DELAY
            (ABSOLUTE
                (IOPATH IC IZ (:1.296:)(:1.211:))
            )
        )
    )

However, as per the format it should be defined as:
(CELL
    (CELLTYPE "GMUX")
    (INSTANCE GMUX)
    (DELAY
        (ABSOLUTE
            (COND IS0_EQ_0 (IOPATH IP IZ (:0.445:)(:0.432:) ) )
            (COND IS0_EQ_1 (IOPATH IC IZ (:1.296:)(:1.211:) ) )
        )
    )
)

lnsharma, 2020-05-08:
In the generated "QLAL4S3B_ss_0p990v_m040c.sdf" file, the LOGIC cell related delays are directly read in from LIB file and the k-factor is not applied. Please apply k-factor while reading in delays from lib file and taking them into sdf.