f4pga/f4pga-arch-defs #1442: OpenTitan support

rw1nkler, 2020-04-24:
Here is the branch with the current progress on OpenTitan support:
https://github.com/antmicro/symbiflow-arch-defs/tree/opentitan_earlgray

Currently, I'm struggling with the error in `vpr_io_place.py`:
```
Generating earlgray_nexys_video/artix7_200t-xc7a200t-virt-xc7a200t-test/top_io.place
Traceback (most recent call last):
  File "/home/build/rwinkler/google-symbiflow-arch-defs/xc/common/utils/prjxray_create_ioplace.py", line 159, in <module>
    main()
  File "/home/build/rwinkler/google-symbiflow-arch-defs/xc/common/utils/prjxray_create_ioplace.py", line 73, in main
    io_place.load_block_names_from_net_file(args.net)
  File "/home/build/rwinkler/google-symbiflow-arch-defs/utils/vpr_io_place.py", line 60, in load_block_names_from_net_file
    "//block[@instance='inpad[0]'] | //block[@instance='outpad[0]']"
  File "src/lxml/etree.pyx", line 1581, in lxml.etree._Element.xpath
  File "src/lxml/xpath.pxi", line 305, in lxml.etree.XPathElementEvaluator.__call__
  File "src/lxml/xpath.pxi", line 225, in lxml.etree._XPathEvaluatorBase._handle_result
lxml.etree.XPathEvalError: Error in xpath expression
xc/xc7/tests/soc/earlgray/CMakeFiles/file_xc_xc7_tests_soc_earlgray_earlgray_nexys_video_artix7_200t-xc7a200t-virt-xc7a200t-test_top_io.place.dir/build.make:71: recipe for target 'xc/xc7/tests/soc/earlgray/earlgray_nexys_video/artix7_200t-xc7a200t-virt-xc7a200t-test/top_io.place' failed
make[3]: *** [xc/xc7/tests/soc/earlgray/earlgray_nexys_video/artix7_200t-xc7a200t-virt-xc7a200t-test/top_io.place] Error 1
make[3]: *** Deleting file 'xc/xc7/tests/soc/earlgray/earlgray_nexys_video/artix7_200t-xc7a200t-virt-xc7a200t-test/top_io.place'
CMakeFiles/Makefile2:222236: recipe for target 'xc/xc7/tests/soc/earlgray/CMakeFiles/file_xc_xc7_tests_soc_earlgray_earlgray_nexys_video_artix7_200t-xc7a200t-virt-xc7a200t-test_top_io.place.dir/all' failed
make[2]: *** [xc/xc7/tests/soc/earlgray/CMakeFiles/file_xc_xc7_tests_soc_earlgray_earlgray_nexys_video_artix7_200t-xc7a200t-virt-xc7a200t-test_top_io.place.dir/all] Error 2
CMakeFiles/Makefile2:222935: recipe for target 'xc/xc7/tests/soc/earlgray/CMakeFiles/earlgray_nexys_video_bit.dir/rule' failed
make[1]: *** [xc/xc7/tests/soc/earlgray/CMakeFiles/earlgray_nexys_video_bit.dir/rule] Error 2
Makefile:2933: recipe for target 'xc/xc7/tests/soc/earlgray/CMakeFiles/earlgray_nexys_video_bit.dir/rule' failed
make: *** [xc/xc7/tests/soc/earlgray/CMakeFiles/earlgray_nexys_video_bit.dir/rule] Error 2
```

However, the same XPath expression works for other designs like button or counter on nexys video. I think that the error is in the wrong `.net` or `.eblif` file itself. Maybe due to unsupported elements of the design or wrong import in previous importing scripts.

mkurc-ant, 2020-04-24:
I've tried to lint the `top.net` xml using `xmllint` and it succeeded. Next, I've tried to query the XPath (part of it) and this is what i got:
```
> xmllint --huge --xpath //block[@instance='inpad[0]'] top.net
XPath error : Memory allocation failed : growing nodeset hit limit

growing nodeset hit limit

^
XPath evaluation failure
```

It seems that the `libxml` has hard time using XPath for such a huge XML file (~330MB). The python wrapper simply misinterprets the error and throws the `lxml.etree.XPathEvalError` exception even though the XPath itself is correct.

The `xmllint` I used is built with:
```
xmllint: using libxml version 20904
   compiled with: Threads Tree Output Push Reader Patterns Writer SAXv1 FTP HTTP DTDValid HTML Legacy C14N Catalog XPath XPointer XInclude Iconv ICU ISO8859X Unicode Regexps Automata Expr Schemas Schematron Modules Debug Zlib Lzma
```
I need to double check that against the version of `libxml` that we use in SymbiFlow.

@rw1nkler For now try applying this patch to `utils/vpr_io_place.py`, this one works by querying all blocks and filtering them in python:
```
diff --git a/utils/vpr_io_place.py b/utils/vpr_io_place.py
index c31fbd7d..5da0e2d8 100644
--- a/utils/vpr_io_place.py
+++ b/utils/vpr_io_place.py
@@ -56,9 +56,11 @@ class IoPlace(object):
         net_root = net_xml.getroot()
         self.net_to_block = {}

-        for block in net_root.xpath(
-                "//block[@instance='inpad[0]'] | //block[@instance='outpad[0]']"
-        ):
+        for block in net_root.xpath("//block"):
+            instance = block.attrib["instance"]
+            if instance != "inpad[0]" and instance != "outpad[0]":
+                continue
+
             top_block = block.getparent()
             assert top_block is not None
             while top_block.getparent() is not net_root:
```

rw1nkler, 2020-05-05:
# Utilization report

Here you can find vivado utilization report:
[top_earlgrey_nexysvideo_utilization_placed.log](https://github.com/SymbiFlow/symbiflow-arch-defs/files/4581283/top_earlgrey_nexysvideo_utilization_placed.log)

# OpenTitan burndown list

Here is the short burndown list for OpenTitan:
- [ ]  Fixes for VPR placement, should be merged. Without them, the placer spreads the blocks all around the device (Merge https://github.com/SymbiFlow/vtr-verilog-to-routing/pull/483)
- [ ]  Add tests and merge BUFGMUX techmap (https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1419)
- [ ] Add tests and merge BUFGCE techmap (https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1418)
- [ ] Create PR with XPath fix needed for parsing `.net` file
- [ ] Generate fasm and analyze the bits
- [ ] Test the bitstream once we have it
- [ ] Create PR with the valid source code with the list of custom changes added to the design 

It will be good to optimize lookahead generation. Currently, it requires more than 10h to complete the computation. This slows down the working process since every change in architecture or in the tools requires lookahead recomputing.

# Changes in the OpenTitan design

Here is the list of changes introduced for OpenTitan:
1. Due to the fact that tristate buffers are not inferred correctly when instantiated in submodules, the top module was changed. (https://github.com/YosysHQ/yosys/issues/1737). Here you can find the changed file:
   - [top_earlgrey_nexysvideo.v](https://github.com/antmicro/symbiflow-arch-defs/blob/opentitan_earlgray/xc/xc7/tests/soc/earlgray/top_earlgrey_nexysvideo.v) (modified)

2. Because of bad RAM36 inference (https://github.com/YosysHQ/yosys/issues/1748), the appropriate workaround was created. It consists of two files:
   - [prim_ram_2p.v](https://github.com/antmicro/symbiflow-arch-defs/blob/opentitan_earlgray/xc/xc7/tests/soc/earlgray/prim_ram_2p.v) (modified)
   - [prim_xilinx_ram_1p](https://github.com/antmicro/symbiflow-arch-defs/blob/opentitan_earlgray/xc/xc7/tests/soc/earlgray/prim_xilinx_ram_1p.v) (created)

3. One `hmac.sv` file after `sv2v` conversion produced the following error in Yosys: `ERROR: 2nd expression of generate for-loop is not constant!`. Due to that, it was changed:
   -  [hmac.sv](https://github.com/antmicro/symbiflow-arch-defs/blob/opentitan_earlgray/xc/xc7/tests/soc/earlgray/hmac.v) (modified)

4. Some parts of Verilog syntax are not supported by Yosys, due to this fact the following commands were used to all the files:

```
if [ "$1" = "prim_lfsr.v" ]; then
	sed -i 's/sv2v_cast_64\((["A-Za-z0-9_]*)\)/\1/g' $1
fi

sed -i 's/parameter unsigned/parameter/g' $1   // $1 is a Verilog file - output of sv2v
sed -i 's/localparam unsigned/localparam/g' $1 
sed -i 's/if (.*) ;//g' $1 
sed -i 's/(strong0, strong1)//g' $1
sed -i 's/(pull0, pull1)//g' $1
sed -i 's/(highz0, weak1)//g' $1 
sed -i 's/(weak0, highz1)//g' $1
sed -i 's/(weak0, weak1)//g' $1
```
Soon I will update the issue with the diff file with all the introduced changes.

# Current work

Currently, we are in the step of `.route` file generation. This step consumes a lot of RAM, and runners with 126Gb of memory were unable to handle it. After introducing the `--congested_routing_iteration_threshold 0.08` flag to the routing process it seems that the routing step does not exceed available resources.

rw1nkler, 2020-05-12:
Currently, the OpenTitan design can be routed with the changes introduced in:
- https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1418
- https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1419
- https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1483

However, the routing process consumes about 90Gb of RAM.

The SymbiFlow failes on the frames step with the following error:
```
  File "/home/build/rwinkler/symbiflow-arch-defs/third_party/prjxray/utils/fasm2frames.py", line 306, in <module>
    main()
  File "/home/build/rwinkler/symbiflow-arch-defs/third_party/prjxray/utils/fasm2frames.py", line 302, in main
    emit_pudc_b_pullup=args.emit_pudc_b_pullup)
  File "/home/build/rwinkler/symbiflow-arch-defs/third_party/prjxray/utils/fasm2frames.py", line 179, in run
    assembler.parse_fasm_filename(filename_in, extra_features=extra_features)
  File "/home/build/rwinkler/symbiflow-arch-defs/third_party/prjxray/prjxray/fasm_assembler.py", line 180, in parse_fasm_filename
    raise FasmLookupError('\n'.join(missing_features))
prjxray.fasm_assembler.FasmLookupError: Segment DB GTP_COMMON_MID_RIGHT, key GTP_COMMON_MID_RIGHT.HCLK_GTP_CK_IN13.HCLK_GTP_CK_MUX13 not found from line 'GTP_COMMON_MID_RIGHT_X167Y23.HCLK_GTP_CK_IN13.HCLK_GTP_CK_MUX13'
Segment DB CLK_HROW_BOT_R, key CLK_HROW_BOT_R.CLK_HROW_BOT_R_CK_BUFG_CASCO22.CLK_HROW_BOT_R_CK_BUFG_CASCIN22 not found from line 'CLK_HROW_BOT_R_X139Y78.CLK_HROW_BOT_R_CK_BUFG_CASCO22.CLK_HROW_BOT_R_CK_BUFG_CASCIN22'
Segment DB CLK_HROW_BOT_R, key CLK_HROW_BOT_R.CLK_HROW_BOT_R_CK_BUFG_CASCO22.CLK_HROW_BOT_R_CK_BUFG_CASCIN22 not found from line 'CLK_HROW_BOT_R_X139Y130.CLK_HROW_BOT_R_CK_BUFG_CASCO22.CLK_HROW_BOT_R_CK_BUFG_CASCIN22'
Segment DB GTP_COMMON_MID_RIGHT, key GTP_COMMON_MID_RIGHT.HCLK_GTP_CK_IN10.HCLK_GTP_CK_MUX10 not found from line 'GTP_COMMON_MID_RIGHT_X167Y23.HCLK_GTP_CK_IN10.HCLK_GTP_CK_MUX10'
Segment DB CLK_HROW_BOT_R, key CLK_HROW_BOT_R.CLK_HROW_BOT_R_CK_BUFG_CASCO20.CLK_HROW_BOT_R_CK_BUFG_CASCIN20 not found from line 'CLK_HROW_BOT_R_X139Y78.CLK_HROW_BOT_R_CK_BUFG_CASCO20.CLK_HROW_BOT_R_CK_BUFG_CASCIN20'
Segment DB CLK_HROW_BOT_R, key CLK_HROW_BOT_R.CLK_HROW_BOT_R_CK_BUFG_CASCO20.CLK_HROW_BOT_R_CK_BUFG_CASCIN20 not found from line 'CLK_HROW_BOT_R_X139Y130.CLK_HROW_BOT_R_CK_BUFG_CASCO20.CLK_HROW_BOT_R_CK_BUFG_CASCIN20'
Segment DB GTP_COMMON_MID_RIGHT, key GTP_COMMON_MID_RIGHT.HCLK_GTP_CK_IN12.HCLK_GTP_CK_MUX12 not found from line 'GTP_COMMON_MID_RIGHT_X167Y23.HCLK_GTP_CK_IN12.HCLK_GTP_CK_MUX12'
Segment DB CLK_HROW_BOT_R, key CLK_HROW_BOT_R.CLK_HROW_BOT_R_CK_BUFG_CASCO2.CLK_HROW_BOT_R_CK_BUFG_CASCIN2 not found from line 'CLK_HROW_BOT_R_X139Y78.CLK_HROW_BOT_R_CK_BUFG_CASCO2.CLK_HROW_BOT_R_CK_BUFG_CASCIN2'
Segment DB CLK_HROW_BOT_R, key CLK_HROW_BOT_R.CLK_HROW_BOT_R_CK_BUFG_CASCO2.CLK_HROW_BOT_R_CK_BUFG_CASCIN2 not found from line 'CLK_HROW_BOT_R_X139Y130.CLK_HROW_BOT_R_CK_BUFG_CASCO2.CLK_HROW_BOT_R_CK_BUFG_CASCIN2'
xc/xc7/tests/soc/earlgray/CMakeFiles/earlgray_nexys_video_bin.dir/build.make:66: recipe for target 'xc/xc7/tests/soc/earlgray/earlgray_nexys_video/artix7_200t-xc7a200t-virt-xc7a200t-test/top.frames' failed
```

tcal-x, 2020-05-29:
I encountered the same error in libxml described above, "Error in xpath expression", at the same place in vpr_io_place.py.   It was a large design -- a version of scalable_proc with N=100.

I ran into the same xml issue also in vpr_place_constraints.py at line 48:
for attr in net_root.xpath("//attribute[@name='LOC']"):

rw1nkler, 2020-07-28:
# Routing issues

After reducing the design frequency to 25MHz, I was able to generate bitstream of EarlGrey design that works on hardware. I used old `symbiflow-arch-defs` version because the routing process on the current master fails due to huge RAM consumption that exceeds 128GB. 

The bitstream was successfully generated using the `symbiflow-arch-defs` in version (f8aa52f3) with conda VPR.
I figured out that there are small differences between locally build VPR and the conda one, but the final routing parameters do not change dramatically - about  `+2ns` for critical path delay. I guess that this is normal behavior.

Here is the table with information about the router iterations:

```
163582 ---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
163583 Iter   Time    pres  BBs    Heap  Re-Rtd  Re-Rtd Overused RR Nodes      Wirelength      CPD       sTNS       sWNS       hTNS       hWNS Est Succ
163584       (sec)     fac Updt    push    Nets   Conns                                       (ns)       (ns)       (ns)       (ns)       (ns)     Iter
163585 ---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
163626    1  625.8     0.0    0 4.0e+09   80416  292570  184703 ( 1.535%) 5882047 (22.8%)   16.386 -2.546e+04    -16.386      0.000      0.000      N/A
163649    2  162.1     4.0   80 7.1e+08   45992  224593   31777 ( 0.264%) 6793439 (26.3%)   16.355 -1.091e+04    -16.355      0.000      0.000      N/A
163650    3   80.3     5.2  315 3.3e+08   19372  102002   16537 ( 0.137%) 7068850 (27.4%)   16.488     -9147.    -16.488      0.000      0.000      N/A
163651    4   62.0     6.8  412 2.5e+08   10355   62812    7694 ( 0.064%) 7266967 (28.2%)   16.440     -9295.    -16.440      0.000      0.000      N/A
163652    5   38.9     8.8  427 1.5e+08    5599   33620    3621 ( 0.030%) 7402195 (28.7%)   16.488     -9200.    -16.488      0.000      0.000      N/A
163653    6   29.0    11.4  349 1.1e+08    3008   16253    1678 ( 0.014%) 7492661 (29.0%)   16.486     -9268.    -16.486      0.000      0.000      N/A
163654    7   27.4    14.9  245 9.9e+07    1551    7943     724 ( 0.006%) 7550901 (29.3%)   16.492     -9245.    -16.492      0.000      0.000      N/A
163655    8   31.9    19.3  149 1.1e+08     783    3327     367 ( 0.003%) 7584782 (29.4%)   16.584     -9258.    -16.584      0.000      0.000      N/A
163656    9   52.9    25.1   81 1.7e+08     467    1415     197 ( 0.002%) 7602250 (29.5%)   16.584     -9259.    -16.584      0.000      0.000      N/A
163657   10   58.3    32.6   50 1.8e+08     273     826     121 ( 0.001%) 7614198 (29.5%)   16.584     -9259.    -16.584      0.000      0.000       16
163658   11   80.2    42.4   33 2.3e+08     185     516      81 ( 0.001%) 7620016 (29.5%)   16.584     -9259.    -16.584      0.000      0.000       17
163659   12   80.7    55.1   17 2.2e+08     123     307      56 ( 0.000%) 7623996 (29.6%)   16.584     -9259.    -16.584      0.000      0.000       18
163660   13   77.0    71.7   13 2.0e+08      86     225      35 ( 0.000%) 7626631 (29.6%)   16.584     -9259.    -16.584      0.000      0.000       20
163661   14   50.5    93.2    9 1.3e+08      55     137      26 ( 0.000%) 7628556 (29.6%)   16.584     -9260.    -16.584      0.000      0.000       20
163662   15   36.9   121.1    4 1.0e+08      40      85      17 ( 0.000%) 7629406 (29.6%)   16.584     -9260.    -16.584      0.000      0.000       21
163663   16   30.1   157.5    4 7.8e+07      29      60      13 ( 0.000%) 7629943 (29.6%)   16.584     -9260.    -16.584      0.000      0.000       22
163664   17   18.0   204.7    3 4.5e+07      19      43      11 ( 0.000%) 7630729 (29.6%)   16.584     -9260.    -16.584      0.000      0.000       22
163665   18   26.7   266.2    3 6.6e+07      19      23      11 ( 0.000%) 7631235 (29.6%)   16.584     -9260.    -16.584      0.000      0.000       23
163666   19   28.3   346.0    2 7.3e+07      22      57      11 ( 0.000%) 7632287 (29.6%)   16.584     -9260.    -16.584      0.000      0.000       24
163667   20   26.5   449.8    3 6.4e+07      18      25       8 ( 0.000%) 7632506 (29.6%)   16.584     -9260.    -16.584      0.000      0.000       26
163668   21   19.2   584.8    0 4.5e+07      12      13       6 ( 0.000%) 7632692 (29.6%)   16.584     -9260.    -16.584      0.000      0.000       28
163669   22   18.6   760.2    0 4.3e+07      11      24       3 ( 0.000%) 7633101 (29.6%)   16.584     -9260.    -16.584      0.000      0.000       28
163670   23    6.8   988.3    0 1.5e+07       5       8       3 ( 0.000%) 7633239 (29.6%)   16.584     -9260.    -16.584      0.000      0.000       28
163671   24    6.8  1284.7    0 1.6e+07       6      22       2 ( 0.000%) 7633457 (29.6%)   16.584     -9260.    -16.584      0.000      0.000       28
163672   25   24.4  1670.2    1 5.1e+07       3       5       1 ( 0.000%) 7633469 (29.6%)   16.584     -9260.    -16.584      0.000      0.000       28
163673   26    2.2  2171.2    0 2110351       3       8       1 ( 0.000%) 7633490 (29.6%)   16.584     -9260.    -16.584      0.000      0.000       27
163674   27    1.7  2822.6    0  489410       1       4       1 ( 0.000%) 7633456 (29.6%)   16.584     -9260.    -16.584      0.000      0.000       27
[... cut ...]
All VPR runs for EarlGrey design consumes a lot of RAM memory. Part of that is related to the size of the routing graph:
```
# Create Device took 32.56 seconds (max_rss 78276.3 MiB, delta_rss +0.0 MiB)
```

Nevertheless, a huge amount of RAM (~80GB) and time (~43min) is used for the `Clean circuit` step:
```
# Clean circuit took 2521.69 seconds (max_rss 78260.4 MiB, delta_rss +77435.4 MiB)
```

This has a huge impact on the time required for the tests since the `clean circuit` step is executed in every VPR run:
[image]
I believe that all four peaks came from the `clean circuit` step in VPR.

rw1nkler, 2020-07-28:
FYI @litghost

rw1nkler, 2020-07-28:
The source code used for the bitstream generation is available in https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1590.
Previous `symbiflow-arch-defs` versions require also:
- https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1419
- https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1523
- https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1505
- https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1483
- https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1418

and the CLK_HROW_BOT_R bits obtained in:
https://github.com/SymbiFlow/prjxray/pull/1381

litghost, 2020-07-28:
I have an initial fix for the RAM explosion in https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/1454 , which will hopefully be merged upstream.  However, the RAM explosion usuallly accompanies an enormous runtime.  Still working on that

litghost, 2020-07-28:
@rw1nkler If you run with https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1612 rebased on master, does it complete?

rw1nkler, 2020-08-11:
Yes, It works after applying the mentioned changes. Also, the routing is successful.
I rebased your changes on (f0e7b4212544e1d55da776fb7a2ff79117e01454)

I rebased your changes on the current master and waiting for the results.

rw1nkler, 2020-08-18:
The diff_fasm step is currently not passing. One issue related to PLLs has been resolved:
https://github.com/SymbiFlow/symbiflow-xc-fasm2bels/pull/17

The second issue is related to wrong constant nets handling in BUFGMUX in Yosys. The error is caused by manual changes to the code and can be fixed. I’m waiting for the bitstream generation here.

HackerFoo, 2020-09-29:
@rw1nkler Any update on this?

rw1nkler, 2020-09-29:
I believe that the CARRY4 modeling needs to be merged before submitting the PR with the OpenTitan test.
As I said before, the test worked after rebasing the changes on the (previous) master.  Since @litghost is working on CARRY4 modeling, I believe that soon the example will be added in the same way as the current Ibex test

litghost, 2020-09-30:
For a concrete progress update on the CARRY4 fixes:
 1. Is required to be cleaned up and green and merged https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1673
 2. https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1674 needs to go green
 3. https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1674 needs to be split into smaller more focused PRs around the various fixes.

Once https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1674 is green, I believe an OpenTitan PR could be added.  I'm working on iterating #1673 and #1674 and fixes issues as they are found.

litghost, 2020-10-02:
#1673 has been merged, and #1674 is now green.  I've split off some changes from #1674 into #1687 #1688 and #1689.  Once those are green and merged, I'll rebase #1674 and address some review comments, and it should be good to merge.