f4pga/f4pga-arch-defs #1449: Travis failing on master?

mithro, 2020-04-29:
https://github.com/SymbiFlow/symbiflow-arch-defs/runs/612217526
```
Downloading and Extracting Packages
pkg-config-0.29.2    | 122 KB    | ##################################### | 100% 
Preparing transaction: done
Verifying transaction: done
Executing transaction: done
[100%] Built target conda_pkg-config
make[2]: Target 'CMakeFiles/all_conda.dir/all' not remade because of errors.
CMakeFiles/Makefile2:5797: recipe for target 'CMakeFiles/all_conda.dir/rule' failed
make[1]: *** [CMakeFiles/all_conda.dir/rule] Error 2
make[1]: Target 'all_conda' not remade because of errors.
Makefile:677: recipe for target 'all_conda' failed
make: *** [all_conda] Error 2
```

Looks like a bunch of the Kokoro builds are also failing?

mithro, 2020-04-29:
https://github.com/SymbiFlow/symbiflow-arch-defs/commit/bc2709a7567c2aa48eddff3e57dca14377b6e128
```
Some checks were not successful
6 failing and 6 successful checks
```

acomodi, 2020-04-29:
All of the failing checks seem to be infrastructure-related. I think by re-running them they should turn green.

I expect the only check to fail to be the xc7 Vendor Tools  one, failing for known issues (https://github.com/SymbiFlow/symbiflow-arch-defs/issues/1429, https://github.com/SymbiFlow/symbiflow-arch-defs/issues/1400).

mithro, 2020-04-29:
@acomodi We should get the symbiflow-robot to rerun the checks if they fail?

mithro, 2020-04-29:
(It already does this for pull requests IIRC?)

closed 2022-05-18 by umarcor