f4pga/f4pga-arch-defs #1455: Branch : Quicklogic : Issue in the top_bit.v, driving a output ASSP port

rakeshm75, 2020-04-29:
Output port of the ASSP macro instance is tied to 0. WBs_ADR is a ASSP output port which is an input to the FPGA, it cannot be driven from the FPGA (cannot be tied to 0).

.WBs_ADR({1'b0, 1'b0, TILE_X1Y1_WBs_ADR[2], TILE_X1Y1_WBs_ADR[3], TILE_X1Y1_WBs_ADR[4], TILE_X1Y1_WBs_ADR[5], TILE_X1Y1_WBs_ADR[6], TILE_X1Y1_WBs_ADR[7], TILE_X1Y1_WBs_ADR[8], TILE_X1Y1_WBs_ADR[9], TILE_X1Y1_WBs_ADR[10], TILE_X1Y1_WBs_ADR[11], TILE_X1Y1_WBs_ADR[12], TILE_X1Y1_WBs_ADR[13], TILE_X1Y1_WBs_ADR[14], TILE_X1Y1_WBs_ADR[15], TILE_X1Y1_WBs_ADR[16]}),

Attached the top_bit.v, file here.

[design3_bit_v.zip](https://github.com/SymbiFlow/symbiflow-arch-defs/files/4553610/design3_bit_v.zip)

rakeshm75, 2020-05-11:
With the latest code the vector signal mapping is correct but still an output port of the ASSP is tied to 0. WBs_ADR is a ASSP output port which is an input to the FPGA, it cannot be driven from the FPGA (cannot be tied to 0).

.WBs_ADR({TILE_X1Y1_WBs_ADR[16], TILE_X1Y1_WBs_ADR[15], TILE_X1Y1_WBs_ADR[14], TILE_X1Y1_WBs_ADR[13], TILE_X1Y1_WBs_ADR[12], TILE_X1Y1_WBs_ADR[11], TILE_X1Y1_WBs_ADR[10], TILE_X1Y1_WBs_ADR[9], TILE_X1Y1_WBs_ADR[8], TILE_X1Y1_WBs_ADR[7], TILE_X1Y1_WBs_ADR[6], TILE_X1Y1_WBs_ADR[5], TILE_X1Y1_WBs_ADR[4], TILE_X1Y1_WBs_ADR[3], TILE_X1Y1_WBs_ADR[2], 1'b0, 1'b0}),

kgugala, 2020-05-13:
This one is fixed

rakeshm75, 2020-05-18:
@kgugala, Yes, now the output port is not driven by the IP, but it is still wrong as TILE_X1Y1_WBs_ADR[16:2] is tied to WBs_ADR [16:0].
Ideally it should be  .WBs_ADR(TILE_X1Y1_WBs_ADR[16:0])  and not
                                 .WBs_ADR(TILE_X1Y1_WBs_ADR[16:2])         
as now the addresses are tied wrongly, WBs_ADR[14:0] to TILE_X1Y1_WBs_ADR[16:2]