f4pga/f4pga-arch-defs #1467: Quicklogic: make target related to analysis is failing

lnsharma, 2020-05-06:
I am running test cases from quicklogic_testsuite using make target related to analysis like the following:
make design5-ql-chandalar_analysis

I tried running on multiple designs and in all the test cases, this is failing and the last few lines look like: 

Also, please note that the below mentioned 3 files are all of 0 size.

Writing Implementation Netlist: top_post_synthesis.v
Writing Implementation Netlist: top_post_synthesis.blif
Writing Implementation SDF    : top_post_synthesis.sdf
make[3]: *** [quicklogic/tests/quicklogic_testsuite/design5/design5-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/analysis.log] Error 134
make[2]: *** [quicklogic/tests/quicklogic_testsuite/design5/CMakeFiles/design5-ql-chandalar_analysis.dir/all] Error 2
make[1]: *** [quicklogic/tests/quicklogic_testsuite/design5/CMakeFiles/design5-ql-chandalar_analysis.dir/rule] Error 2
make: *** [quicklogic/tests/quicklogic_testsuite/design5/CMakeFiles/design5-ql-chandalar_analysis.dir/rule] Error 2

kgugala, 2020-05-06:
@lnsharma can you attach the `vpr_stdout.log` from this run?

lnsharma, 2020-05-06:
[vpr_stdout.log](https://github.com/SymbiFlow/symbiflow-arch-defs/files/4586507/vpr_stdout.log)
Attaching vpr_stdout.log file for test case run: symbiflow-arch-defs/build/quicklogic/tests/quicklogic_testsuite/design5

mkurc-ant, 2020-05-20:
So the issue is actually caused by a bug in VPR in writing post-synthesis netlist (that is enabled when the `analysis` target is run).

I've issues a PR to the upstream VPR. Once the fix is there we can apply it in SymbiFlow.

mkurc-ant, 2020-05-20:
A PR with new master+wip: https://github.com/SymbiFlow/vtr-verilog-to-routing/pull/495

lnsharma, 2020-05-21:
Are you planning to merge this into branch antmicro/quicklogic-upstream-rebase? I can them merge it to quicklogic-corp/quicklogic-upstream-rebase as this is the branch we are actively using.

mkurc-ant, 2020-05-21:
Either that or I'll create a separate PR so you can review it.