f4pga/f4pga-arch-defs #1468: Add support for MMCM

mkurc-ant, 2020-05-06:
Xilinx 7-series FPGAs have PLLs (PLLE2_ADV) and MMCMs (MMCME2_ADV). Both have the same function though MMCMs have some additional features.

Currently SymbiFlow has support only for PLLs. Adding support for MMCMs would require duplicating and augmenting what's done for PLLs. All bits for MMCMs should be known but it has to be double- checked.

litghost, 2020-10-02:
PR opened for unknown bits: https://github.com/SymbiFlow/prjxray/pull/1447

arch-def side of this is still needed.

mkurc-ant, 2020-10-27:
A WIP PR to arch-defs: https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1729

mkurc-ant, 2021-08-23:
https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1729 has been merged. Closing

closed 2021-08-23 by mkurc-ant