f4pga/f4pga-arch-defs #1489: VPR route design through unknown GTP PIPs

rw1nkler, 2020-05-15:
VPR is routing some designs on artix200t through unknown GTP PIPs. 
In those designs, the GTP tiles carry the horizontal clocks to the BUFHCE primitives. Since the interconnect points are "real" and no "pseudo" PIPs, the bits corresponding to the should be known. However, currently, we do not support them.

Because of the missing fasm features, the bitstream generation for these designs cannot be finished. This is the case for i.e., OpenTitan design.

This is an example of ane error which shows unsupported fasm features:
```
Traceback (most recent call last):
  File "/home/build/rwinkler/google-symbiflow-arch-defs/third_party/prjxray/utils/fasm2frames.py", line 306, in <module>
    main()
  File "/home/build/rwinkler/google-symbiflow-arch-defs/third_party/prjxray/utils/fasm2frames.py", line 302, in main
    emit_pudc_b_pullup=args.emit_pudc_b_pullup)
  File "/home/build/rwinkler/google-symbiflow-arch-defs/third_party/prjxray/utils/fasm2frames.py", line 179, in run
    assembler.parse_fasm_filename(filename_in, extra_features=extra_features)
  File "/home/build/rwinkler/google-symbiflow-arch-defs/third_party/prjxray/prjxray/fasm_assembler.py", line 180, in parse_fasm_filename
    raise FasmLookupError('\n'.join(missing_features))
prjxray.fasm_assembler.FasmLookupError: Segment DB GTP_COMMON_MID_RIGHT, key GTP_COMMON_MID_RIGHT.HCLK_GTP_CK_IN13.HCLK_GTP_CK_MUX13 not found from line 'GTP_COMMON_MID_RIGHT_X167Y23.HCLK_GTP_CK_IN13.HCLK_GTP_CK_MUX13'
Segment DB CLK_HROW_BOT_R, key CLK_HROW_BOT_R.CLK_HROW_BOT_R_CK_BUFG_CASCO22.CLK_HROW_BOT_R_CK_BUFG_CASCIN22 not found from line 'CLK_HROW_BOT_R_X139Y78.CLK_HROW_BOT_R_CK_BUFG_CASCO22.CLK_HROW_BOT_R_CK_BUFG_CASCIN22'
Segment DB CLK_HROW_BOT_R, key CLK_HROW_BOT_R.CLK_HROW_BOT_R_CK_BUFG_CASCO22.CLK_HROW_BOT_R_CK_BUFG_CASCIN22 not found from line 'CLK_HROW_BOT_R_X139Y130.CLK_HROW_BOT_R_CK_BUFG_CASCO22.CLK_HROW_BOT_R_CK_BUFG_CASCIN22'
Segment DB GTP_COMMON_MID_RIGHT, key GTP_COMMON_MID_RIGHT.HCLK_GTP_CK_IN10.HCLK_GTP_CK_MUX10 not found from line 'GTP_COMMON_MID_RIGHT_X167Y23.HCLK_GTP_CK_IN10.HCLK_GTP_CK_MUX10'
Segment DB CLK_HROW_BOT_R, key CLK_HROW_BOT_R.CLK_HROW_BOT_R_CK_BUFG_CASCO20.CLK_HROW_BOT_R_CK_BUFG_CASCIN20 not found from line 'CLK_HROW_BOT_R_X139Y78.CLK_HROW_BOT_R_CK_BUFG_CASCO20.CLK_HROW_BOT_R_CK_BUFG_CASCIN20'
Segment DB CLK_HROW_BOT_R, key CLK_HROW_BOT_R.CLK_HROW_BOT_R_CK_BUFG_CASCO20.CLK_HROW_BOT_R_CK_BUFG_CASCIN20 not found from line 'CLK_HROW_BOT_R_X139Y130.CLK_HROW_BOT_R_CK_BUFG_CASCO20.CLK_HROW_BOT_R_CK_BUFG_CASCIN20'
Segment DB GTP_COMMON_MID_RIGHT, key GTP_COMMON_MID_RIGHT.HCLK_GTP_CK_IN12.HCLK_GTP_CK_MUX12 not found from line 'GTP_COMMON_MID_RIGHT_X167Y23.HCLK_GTP_CK_IN12.HCLK_GTP_CK_MUX12'
Segment DB CLK_HROW_BOT_R, key CLK_HROW_BOT_R.CLK_HROW_BOT_R_CK_BUFG_CASCO2.CLK_HROW_BOT_R_CK_BUFG_CASCIN2 not found from line 'CLK_HROW_BOT_R_X139Y78.CLK_HROW_BOT_R_CK_BUFG_CASCO2.CLK_HROW_BOT_R_CK_BUFG_CASCIN2'
Segment DB CLK_HROW_BOT_R, key CLK_HROW_BOT_R.CLK_HROW_BOT_R_CK_BUFG_CASCO2.CLK_HROW_BOT_R_CK_BUFG_CASCIN2 not found from line 'CLK_HROW_BOT_R_X139Y130.CLK_HROW_BOT_R_CK_BUFG_CASCO2.CLK_HROW_BOT_R_CK_BUFG_CASCIN2'
xc/xc7/tests/soc/earlgray/CMakeFiles/earlgray_nexys_video_bin.dir/build.make:66: recipe for target 'xc/xc7/tests/soc/earlgray/earlgray_nexys_video/artix7_200t-xc7a200t-virt-xc7a200t-test/top.frames' failed
```
Possible solutions:
- reduce the grid to exclude GTP options
- do not import unwanted connections by modifying RR graph

mithro, 2020-05-19:
@litghost - What is you thoughts here?

mithro, 2020-05-19:
@mkurc-ant Had some screenshots?

rw1nkler, 2020-05-19:
[image]
[image]
[image]

mkurc-ant, 2020-05-19:
These are those screenshots. If you look closely at the PIPs they do not have source wires.

litghost, 2020-05-19:
> These are those screenshots. If you look closely at the PIPs they do not have source wires.

That is super weird!  What is the source for those nets?