f4pga/f4pga-arch-defs #1494: Missing CLK_HROW_BOT_R bits

rw1nkler, 2020-05-19:
These bits are missing in the current artix7 database. However, they can be found in kintex7 database:
```
CLK_HROW_BOT_R_CK_BUFG_CASCO22.CLK_HROW_BOT_R_CK_BUFG_CASCIN22
CLK_HROW_BOT_R_CK_BUFG_CASCO20.CLK_HROW_BOT_R_CK_BUFG_CASCIN20
CLK_HROW_BOT_R_CK_BUFG_CASCO2.CLK_HROW_BOT_R_CK_BUFG_CASCIN2
```

litghost, 2020-05-19:
This is because the clock network fuzzing on the 50T fabric doesn't have a bottom clock cascade.  One solution could be to switch the main part from the 50T fabric to 100T or 200T fabrics, which do have bottom clock cascades.

rw1nkler, 2020-05-26:
I was able to generate those bits using A100T settings. For A200T I have the following error during fuzzing:
```
ERROR: [DRC PDIL-1] Invalid Site Configuration: Invalid configuration for site BUFGCTRL_X0Y1. Reason: Site pin to site pin route-thru requires conflicting attribute enum values for user logic element 'BUFGCTRL in site 'BUFGCTRL_X0Y1'. Att
ribute 'PRESELECT_I0' is programmed to 'FALSE' but needs a value of 'TRUE' for the route-thru.
.
ERROR: [DRC RTSTAT-2] Partially routed nets: 6 net(s) are partially routed. The problem bus(es) and/or net(s) are lut_6_o, lut_7_o, lut_8_o, lut_9_o, lut_10_o, and lut_11_o.
ERROR: [DRC RTSTAT-5] Partial antennas: 7 net(s) have a partial antenna. The problem bus(es) and/or net(s) are lut_6_o, lut_7_o, lut_8_o, lut_9_o, lut_10_o, lut_11_o, and mmcm_clock_MMCME2_ADV_X1Y2_0.
ERROR: [DRC RTSTAT-11] Invalid Site Programming: Invalid site programming for net O_BUFR_X1Y3. The following site(s) are invalid: BUFGCTRL_X0Y1.
ERROR: [DRC RTSTAT-11] Invalid Site Programming: Invalid site programming for net lut_10_o. The following site(s) are invalid: BUFGCTRL_X0Y1.
ERROR: [DRC RTSTAT-11] Invalid Site Programming: Invalid site programming for net lut_11_o. The following site(s) are invalid: BUFGCTRL_X0Y1.
ERROR: [DRC RTSTAT-11] Invalid Site Programming: Invalid site programming for net lut_6_o. The following site(s) are invalid: BUFGCTRL_X0Y1.
ERROR: [DRC RTSTAT-11] Invalid Site Programming: Invalid site programming for net lut_7_o. The following site(s) are invalid: BUFGCTRL_X0Y1.
ERROR: [DRC RTSTAT-11] Invalid Site Programming: Invalid site programming for net lut_8_o. The following site(s) are invalid: BUFGCTRL_X0Y1.
ERROR: [DRC RTSTAT-11] Invalid Site Programming: Invalid site programming for net lut_9_o. The following site(s) are invalid: BUFGCTRL_X0Y1.
ERROR: [DRC RTSTAT-11] Invalid Site Programming: Invalid site programming for net mmcm_clock_MMCME2_ADV_X1Y2_0. The following site(s) are invalid: BUFGCTRL_X0Y1.
```

rw1nkler, 2020-08-12:
This issue has been resolved by https://github.com/SymbiFlow/prjxray/issues/1382

closed 2020-08-12 by rw1nkler