f4pga/f4pga-arch-defs #1496: Changes in miniconda version generate conda env issues

acomodi, 2020-05-20:
With the latest master I get the following error when running `make format_python` (with a freshly cloned repo):

```
yapf failed to format /data/tmp/symbiflow-arch-defs/xc/xc7/tests/common/read_uart.py
../common/python_health_helper.sh: line 12: /data/tmp/symbiflow-arch-defs/build/env/conda/bin/yapf: No such file or directory
yapf failed to format /data/tmp/symbiflow-arch-defs/xc/xc7/tests/ddr/scripts/arty.py
../common/python_health_helper.sh: line 12: /data/tmp/symbiflow-arch-defs/build/env/conda/bin/yapf: No such file or directory
yapf failed to format /data/tmp/symbiflow-arch-defs/xc/xc7/tests/ddr/scripts/sdram_init.py
../common/python_health_helper.sh: line 12: /data/tmp/symbiflow-arch-defs/build/env/conda/bin/yapf: No such file or directory
yapf failed to format /data/tmp/symbiflow-arch-defs/xc/xc7/tests/ddr/scripts/test_sdram.py
../common/python_health_helper.sh: line 12: /data/tmp/symbiflow-arch-defs/build/env/conda/bin/yapf: No such file or directory
yapf failed to format /data/tmp/symbiflow-arch-defs/xc/xc7/tests/ff_sr_ce/create_sby.py
../common/python_health_helper.sh: line 12: /data/tmp/symbiflow-arch-defs/build/env/conda/bin/yapf: No such file or directory
yapf failed to format /data/tmp/symbiflow-arch-defs/xc/xc7/tests/ff_sr_ce/generate.py
../common/python_health_helper.sh: line 12: /data/tmp/symbiflow-arch-defs/build/env/conda/bin/yapf: No such file or directory
yapf failed to format /data/tmp/symbiflow-arch-defs/xc/xc7/tests/iosettings/generate.py
../common/python_health_helper.sh: line 12: /data/tmp/symbiflow-arch-defs/build/env/conda/bin/yapf: No such file or directory
yapf failed to format /data/tmp/symbiflow-arch-defs/xc/xc7/tests/ps7/axi_lite_reg/build.py
../common/python_health_helper.sh: line 12: /data/tmp/symbiflow-arch-defs/build/env/conda/bin/yapf: No such file or directory
yapf failed to format /data/tmp/symbiflow-arch-defs/xc/xc7/tests/serdes/generate_tests.py
../common/python_health_helper.sh: line 12: /data/tmp/symbiflow-arch-defs/build/env/conda/bin/yapf: No such file or directory
yapf failed to format /data/tmp/symbiflow-arch-defs/xc/xc7/tests/soc/litex/mini/scripts/minilitex_arty.py
../common/python_health_helper.sh: line 12: /data/tmp/symbiflow-arch-defs/build/env/conda/bin/yapf: No such file or directory

```

System info:
```
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.3 LTS
Release:        18.04
Codename:       bionic
```

litghost, 2020-05-20:
Did you run the all_conda target?

acomodi, 2020-05-20:
> Did you run the all_conda target?

Yes:
```
make env
make all_conda
make format_python
```

litghost, 2020-05-20:
Travis and Kokoro are both green.   Travis is running xenial (16.06).  Are you on a 32 or 64 bit host?

litghost, 2020-05-20:
I suspect you are missing the 32-bit glib libraries, and ldd is failing on yapf.

acomodi, 2020-05-21:
This is actually a local problem in my machine. After a little debugging, I found out that the some packages are getting installed in another conda environment I have in the system (regardless of the Miniconda version). Closing

closed 2020-05-21 by acomodi