f4pga/f4pga-arch-defs #1537: Branch : Quicklogic : Failed to find matching architecture model for 'mux4x0'

rakeshm75, 2020-06-22:
When running Symbliflow on the attached design, I am getting the following error:

# Building complex block graph took 0.04 seconds (max_rss 25.9 MiB, delta_rss +13.1 MiB)
Error 1: 
Type: Blif file
File: /adhome/rakeshm/symbiflow-arch-defs/build/quicklogic/tests/quicklogic_testsuite/design2/design2-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.eblif
Line: 7622
Message: Failed to find matching architecture model for 'mux4x0'

# Load circuit
# Load circuit took 0.10 seconds (max_rss 31.0 MiB, delta_rss +5.1 MiB)
The entire flow of VPR took 0.31 seconds (max_rss 31.0 MiB)
make[3]: *** [quicklogic/tests/quicklogic_testsuite/design2/design2-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.net] Error 1
make[2]: *** [quicklogic/tests/quicklogic_testsuite/design2/CMakeFiles/file_quicklogic_tests_quicklogic_testsuite_design2_design2-ql-chandalar_ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp_top.net.dir/all] Error 2
make[1]: *** [quicklogic/tests/quicklogic_testsuite/design2/CMakeFiles/design2-ql-chandalar_analysis.dir/rule] Error 2
make: *** [quicklogic/tests/quicklogic_testsuite/design2/CMakeFiles/design2-ql-chandalar_analysis.dir/rule] Error 2

[design2.zip](https://github.com/SymbiFlow/symbiflow-arch-defs/files/4811921/design2.zip)

kkumar23, 2020-06-29:
The issue still exists for the designs quicklogic_testsuit/Cavlc_top , rs_decoder_1

mkurc-ant, 2020-06-29:
@kkumar23 On which exact commit of Yosys you see it failing? It should work correctly from Yosys from `cac5aa3` and later.

kkumar23, 2020-06-29:
@mkurc-ant : we are seeing on the Quicklogic yosys branch.

mkurc-ant, 2020-07-01:
@kkumar23 Should be fixed by now as the commit that fixed the techmap for `mux4x0` has been integrated into the `QuickLogic-Corp:quicklogic-rebased` branch.