f4pga/f4pga-arch-defs #1542: prjxray_create_place_constraints.py fails for unconstrained PLLE2_ADV on xc7a100T device

tcal-x, 2020-06-23:
Error seen:
```
Traceback (most recent call last):
  File "/home/tcal/2nd-tcal-x/symbiflow-arch-defs/xc/common/utils/prjxray_create_place_constraints.py", line 693, in <module>
    main()
  File "/home/tcal/2nd-tcal-x/symbiflow-arch-defs/xc/common/utils/prjxray_create_place_constraints.py", line 681, in main
    vpr_grid, loc_in_use, block_locs, blocks, grid_capacities):
  File "/home/tcal/2nd-tcal-x/symbiflow-arch-defs/xc/common/utils/prjxray_create_place_constraints.py", line 514, in place_clocks
    for potential_loc in sorted(available_placements[key]):
KeyError: ('PLLE2_ADV', 5)
```

To reproduce:
* Sync to current head of master (a8c60d95).
* Rerun "make env" to rerun cmake to get the new 100T test targets.
* Edit the file `xc/xc7/tests/soc/litex/mini_ddr/minilitex_ddr_arty100t.v` as follows:
  * Delete or comment out the line `(* LOC="PLLE2_ADV_X1Y1" *)`
* `cd` to `build/xc/xc7/tests/soc/litex/mini_ddr`.
* Run `make minilitex_ddr_arty100t_bin` to produce the error.
  * This will take several hours the first time, so run overnight.
  * It takes less than a minute to get to the error in subsequent runs.

Notes:
* The original 35T design constrained the IDELAYCTRL to X1Y0.  With the 100T it is constrained to X1Y1.
* In the original design, the PLLE2_ADV did not need to be constrained.   With 100T, if it is not constrained, we encounter the script error above.   If it _is_ constrained to X1Y1, then everything works.

This is not blocking, since there is the workaround of adding the LOC constraint to the PLLE2_ADV.

tcal-x, 2020-07-17:
Hi @litghost , who is the best person to own this?   It involves the constraint solver in the clock placement script.   This should be  a pretty easy problem to solve, but I can't find anything obvious, such as a hardcoding of clock region X1Y0 (which wouldn't work on 100T)

litghost, 2020-07-17:
> Hi @litghost , who is the best person to own this? It involves the constraint solver in the clock placement script. This should be a pretty easy problem to solve, but I can't find anything obvious, such as a hardcoding of clock region X1Y0 (which wouldn't work on 100T)

I believe this issue is on the list of bugs to fix for useability reasons, so it will get resolved.  Other issues are being actively worked on, and I don't see this as a super high priority because of the ease of the workaround.  Do you believe it needs a higher priority?

tcal-x, 2020-07-17:
Hi @litghost , that exact point of view occurred to me after I posted my note, so we're in complete agreement.   It will get done eventually, but it's not blocking anyone now, so there's no urgency, and no reason to bump it to higher priority.

litghost, 2020-10-29:
@acomodi I believe this has been fixed?

acomodi, 2020-10-29:
Yes, this has been fixed. Closing

closed 2020-10-29 by acomodi