f4pga/f4pga-arch-defs #1546: fasm2bels: max recursion depth exceeded, make_routes.py, xc7a100t, soc/litex/base

tcal-x, 2020-06-24:
Error seen:
```
[100%] Generating baselitex_arty100t/artix7_100t-xc7a100t-virt-xc7a100t-test/top_bit.v
Traceback (most recent call last):
  File "/home/tcal/2nd-tcal-x/symbiflow-arch-defs/build/env/conda/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/tcal/2nd-tcal-x/symbiflow-arch-defs/build/env/conda/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/tcal/2nd-tcal-x/symbiflow-arch-defs/xc/xc7/fasm2bels/__main__.py", line 4, in <module>
    main()
  File "/home/tcal/2nd-tcal-x/symbiflow-arch-defs/xc/xc7/fasm2bels/fasm2bels.py", line 403, in main
    top.make_routes(allow_orphan_sinks=args.allow_orphan_sinks)
  File "/home/tcal/2nd-tcal-x/symbiflow-arch-defs/xc/xc7/fasm2bels/verilog_modeling.py", line 1502, in make_routes
    net_map=self.net_map,
  File "/home/tcal/2nd-tcal-x/symbiflow-arch-defs/xc/xc7/fasm2bels/make_routes.py", line 709, in make_routes
    allow_orphan_sinks=allow_orphan_sinks
  File "/home/tcal/2nd-tcal-x/symbiflow-arch-defs/xc/xc7/fasm2bels/make_routes.py", line 583, in expand_sink
    allow_orphan_sinks=allow_orphan_sinks
  File "/home/tcal/2nd-tcal-x/symbiflow-arch-defs/xc/xc7/fasm2bels/make_routes.py", line 453, in expand_sink
    allow_orphan_sinks=allow_orphan_sinks
  File "/home/tcal/2nd-tcal-x/symbiflow-arch-defs/xc/xc7/fasm2bels/make_routes.py", line 583, in expand_sink
    allow_orphan_sinks=allow_orphan_sinks
...
  File "/home/tcal/2nd-tcal-x/symbiflow-arch-defs/xc/xc7/fasm2bels/make_routes.py", line 453, in expand_sink
    allow_orphan_sinks=allow_orphan_sinks
  [Previous line repeated 457 more times]
  File "/home/tcal/2nd-tcal-x/symbiflow-arch-defs/xc/xc7/fasm2bels/make_routes.py", line 436, in expand_sink
    sink_node_pkey = get_node_pkey(conn, sink_wire_pkey)
  File "/home/tcal/2nd-tcal-x/symbiflow-arch-defs/xc/xc7/fasm2bels/connection_db_utils.py", line 60, in get_node_pkey
    c = conn.cursor()
RecursionError: maximum recursion depth exceeded while calling a Python object
xc/xc7/tests/soc/litex/base/CMakeFiles/file_xc_xc7_tests_soc_litex_base_baselitex_arty100t_artix7_100t-xc7a100t-virt-xc7a100t-test_top_bit.v.dir/build.make:82: recipe for target 'xc/xc7/tests/soc/litex/base/baselitex_arty100t/artix7_100t-xc7a100t-virt-xc7a100t-test/top_bit.v' failed
```

This test was rebased on top of current master (a8c60d95).

litghost, 2020-06-24:
Is this a case of an infinite loop or a net that was long enough to block to stack?

tcal-x, 2020-06-25:
> Is this a case of an infinite loop or a net that was long enough to block to stack?
> 
> You can attempt to increase the Python stack to quickly check.

Thanks Keith, yep, if I do `sys.setrecursionlimit(1250)`, it finishes without error.   The exact value needed is somewhere between 1200 and 1250.   Should I add it to this script, or is there a different preferred method?

litghost, 2020-06-25:
> Should I add it to this script, or is there a different preferred method?

We should convert from a recursive algorithm, to an explicit stack.  Larger graphs will have longer paths, and limiting the path length based on the stack limit of threads is a bad plan.

mithro, 2020-06-25:
@litghost / @tcal-x -- It seems like we should just set `sys.setrecursionlimit(5000)` and then put converting the recursive algorithm to an explicit stack on the TODO list?

litghost, 2020-06-25:
> @litghost / @tcal-x -- It seems like we should just set `sys.setrecursionlimit(5000)` and then put converting the recursive algorithm to an explicit stack on the TODO list?

This runs the risk of hitting the underlying thread stack.  When that blows, it will cause a SIGSEGV, rather than a python exception.  So that solution is not a good one.

tcal-x, 2020-06-30:
@litghost yeah there must have been something very wrong with vpr.   After rebasing, the error went away on its own, so I added code to check the max recursion depth, and the max that it hit was just **44**.

So I'll close this and not make any code changes.

closed 2020-06-30 by tcal-x