f4pga/f4pga-arch-defs #1550: Support multiple synth IO tiles at the same vpr coordinate

andrewb1999, 2020-06-25:
For custom harnesses and partition regions, there should be the ability to place multiple synth IO tiles at the same VPR coordinate, connecting to different wires at the border of the ROI.

I think this can be implemented relatively easily using by increasing the tile capacity, but I am unsure how to handle an INPAD and and OUTPAD at the same vpr coordinate.  My initial thought is to merge these together into an SYN-IOPAD and maybe have the INPAD and OUTPAD as equivalent sites.

andrewb1999, 2020-06-25:
@litghost Can you let me know your thoughts?

litghost, 2020-06-25:
So @acomodi add the feature to support this, which is heterogeneous tiles.  There are documentation and examples upstream.  I'll find a link.

litghost, 2020-06-26:
Link to doc: https://docs.verilogtorouting.org/en/latest/tutorials/arch/heterogeneous_tiles/#heterogeneous-tiles-tutorial

closed 2020-09-30 by andrewb1999