f4pga/f4pga-arch-defs #1553: Branch Quicklogic : The latest global clock support has issues

kkumar23, 2020-06-26:
With the latest global clock support below issues occur:
1) The clock pins can only be mapped to clock ports and cannot be mapped on BIPADS

2) Designs [quicklogic-testsuit/bin2seven, clock_tre, counter_32bit] show router issues as shown below :
Warning 3094: No routing path for connection to sink_rr 22700, retrying with full device bounding box
Cannot route from TL-CLOCK[0].CLOCK0_IC[0] (RR node: 24874 type: SOURCE location: (20,2) class: 0 capacity: 1) to TL-LOGIC[0].LOGIC0_QRT[0] (RR node: 22700 type: SINK location: (18,18) class: 20 capacity: 1) -- no possible path
Failed to route connection from 'reset' to 'LED_out_LUT4_O_I3_LUT4_O_I0_LUT3_O_I0_LUT4_O_I2_LUT2_I1_O_LUT4_I0.t_lut.t_frag' for net '$auto$clkbufmap.cc:247:execute$242605' (#9)
Routing failed.

3) Designs [quicklogic-testsuit/design1/design5/design6] have the below issue:
File "/home/build/symbiflow-arch-defs/utils/vpr_io_place.py", line 104, in get_top_level_block_instance_for_net
    block_name = self.net_to_block[net_name]
KeyError: 'I2S_DIN_i'

mkurc-ant, 2020-06-26:
We are currently working hard on the solution:

1. Please look at the pull request for the `quicklogic-ios` Yosys plugin: https://github.com/QuickLogic-Corp/yosys-symbiflow-plugins/pull/2. This fixes the issue of constraining a `ckpad` to a `BIDIR`. That fix won't work alone, there is an upcoming PR to symbiflow-arch-defs that will take advantage of it.

mkurc-ant, 2020-06-26:
2. This issue is caused by incorrect representation of the `GMUX` model in VPR architecture. VPR simply places these buffers incorrectly. There is a work-in-progress fix for that.

mkurc-ant, 2020-06-26:
3. We are currently investigating this problem. It looks like a part of the design gets "optimized" either by Yosys of VPR. That is why the script that generates placement constraints reports an error.

mkurc-ant, 2020-06-29:
Status update:

1. The PR https://github.com/QuickLogic-Corp/yosys-symbiflow-plugins/pull/2 has been merged so the issue should be fixes by now
2. There is an open PR that fixes the `GMUX` insertion and placement: https://github.com/QuickLogic-Corp/symbiflow-arch-defs/pull/36
3. Finally, a third PR to symbiflow-arch-defs that prevents Yosys from removing techmapped *ASSP* cell: https://github.com/QuickLogic-Corp/symbiflow-arch-defs/pull/37

mkurc-ant, 2020-07-01:
All the PRs have been merged so I would consider the issue closed.