f4pga/f4pga-arch-defs #1554: Branch Quicklogic : The no of PB-5XGMUX available shows only 1.

kkumar23, 2020-06-26:
The total global muxes available are 5 and not 1.
Refer quicklogic-testsuite/design3:
Message: Failed to find device which satisifies resource requirements required: 
                   PB-LOGIC: 675, PB-ASSP: 1, PB-5XGMUX: 2, PB-SYN_VCC: 1, PB-SYN_GND: 1, PB-RAM: 3, PB-MULT: 1, PB-BIDIR: 4, PB-CLOCK: 0, PB-SDIOMUX: 0
(available PB-LOGIC: 891, PB-ASSP: 1, PB-5XGMUX: 1, PB-SYN_VCC: 1, PB-SYN_GND: 1, PB-RAM: 4, PB-MULT: 2, PB-BIDIR: 32, PB-CLOCK: 5, PB-SDIOMUX: 14)

mkurc-ant, 2020-06-26:
So currently the GMUXes are modeled as a single tile that contains 5 GMUX sites. That tile (actually the top-level pb_type in VPR) is named PB-5XGMUX. So there is one tile with 5 GMUXes that's why VPR reports it as only one.

Unfortunately, the model does not work correctly as there is no control which one of these 5 GMUXes is chosen during packing/placement.

The solution that is being worked on. It includes modeling the GMUX tile as 5 separate tiles with one GMUX site in each. In both cases the total number of GMUX cells is correct. Expect a pull request soon with the fix.

mkurc-ant, 2020-06-29:
The issue should be already fixed as https://github.com/QuickLogic-Corp/symbiflow-arch-defs/pull/34 had been merged.