f4pga/f4pga-arch-defs #1560: Investigate long path in soc/litex/base with xc7a100t: 1200+ length

tcal-x, 2020-06-30:
Follow-on to issue #1546 .

Directory `build/xc/xc7/tests/soc/litex/base`, target `baselitex_arty100t_vivado_diff_fasm`.

During fasm2bels, the `make_routes.py` crashed because the path length exceeded the recursion limit of 1000.  The path length was somewhere between 1200 and 1250.  This will be fixed by increasing the maximum recursion limit.

But why is such a long path needed when the part grid is 90x200?   Perhaps we should dump source/sink info from `make_routes.py` for such long paths.

litghost, 2020-10-29:
I don't believe we've seen this happen again for a while?  Close for now?

tcal-x, 2020-10-29:
Correct, we weren't able to reproduce it since other things had moved ahead.

closed 2020-10-29 by tcal-x