f4pga/f4pga-arch-defs #1580: SRL tests fail if top outputs are connected to GND with new Yosys

acomodi, 2020-07-08:
Top level outputs connected with GND net produce the following error:

```
Initial Net Connection Criticality Histogram:
[        0:      0.1) 249 ( 20.0%) |***********************************************
[      0.1:      0.2)   5 (  0.4%) |*
[      0.2:      0.3)  61 (  4.9%) |************
[      0.3:      0.4) 108 (  8.7%) |********************
[      0.4:      0.5)  46 (  3.7%) |*********
[      0.5:      0.6)  64 (  5.1%) |************
[      0.6:      0.7) 126 ( 10.1%) |************************
[      0.7:      0.8) 218 ( 17.5%) |*****************************************
[      0.8:      0.9) 249 ( 20.0%) |***********************************************
[      0.9:        1) 119 (  9.6%) |**********************
## Initializing router criticalities took 0.05 seconds (max_rss 671.3 MiB, delta_rss +0.0 MiB)
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Iter   Time    pres  BBs    Heap  Re-Rtd  Re-Rtd Overused RR Nodes      Wirelength      CPD       sTNS       sWNS       hTNS       hWNS Est Succ
      (sec)     fac Updt    push    Nets   Conns                                       (ns)       (ns)       (ns)       (ns)       (ns)     Iter
---- ------ ------- ---- ------- ------- ------- ----------------- --------------- -------- ---------- ---------- ---------- ---------- --------
Warning 22: No routing path for connection to sink_rr 475192, retrying with full device bounding box
Warning 23: No routing path for connection to sink_rr 393515, retrying with full device bounding box
Cannot route from SYN-GND[0].GND[0] (RR node: 103530 type: SOURCE location: (30,27) class: 0 capacity: 1) to SYN-OUTPAD[0].outpad[0] (RR node: 393515 type: SINK location: (62,24) class: 0 capacity: 1) -- no possible path
Failed to route connection from '$false' to 'out:led[15]' for net 'led[4]' (#0)
Routing failed.
# Routing took 0.57 seconds (max_rss 671.3 MiB, delta_rss +0.0 MiB)
Circuit is unroutable with a channel width factor of 500.
VPR failed to implement circuit
The entire flow of VPR took 3.00 seconds (max_rss 671.3 MiB)
```

litghost, 2020-07-08:
What's happening here is that yosys is optimizing away the SRL.  Probably need a `(* KEEP *)` or equiv?  Assuming that yosys accepts that.

litghost, 2020-07-08:
Other possible problem is that the test logic has a bug, and as a result yosys is optimizing the entire design away.  It happens :/

litghost, 2020-07-08:
I see the issue, and have a straight forward change that fixes it.

litghost, 2020-07-08:
The issue was that yosys was optimizing something away, but in this case it was a LUT2 that was being used to avoid directly connected unused LED's to VCC/GND.  I've updated the relevant tests with a comment explaining what is going on.  Something we should consider in the future is moving those tests outside the ROI, which would remove the need for having weird logic around GND/VCC directly to pads.