f4pga/f4pga-arch-defs #1588: Branch : Quicklogic : Crash while running analysis on the attached design using logic cell macro

rakeshm75, 2020-07-13:
Get following error when running the attached design (using logic cell macro):

Writing Implementation Netlist: top_post_synthesis.v
Writing Implementation Netlist: top_post_synthesis.blif
Writing Implementation SDF    : top_post_synthesis.sdf
make[3]: *** [quicklogic/pp3/tests/quicklogic_testsuite/test_logic_cell/test_logic_cell-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/analysis.log] Error 134
make[2]: *** [quicklogic/pp3/tests/quicklogic_testsuite/test_logic_cell/CMakeFiles/test_logic_cell-ql-chandalar_analysis.dir/all] Error 2
make[1]: *** [quicklogic/pp3/tests/quicklogic_testsuite/test_logic_cell/CMakeFiles/test_logic_cell-ql-chandalar_analysis.dir/rule] Error 2
make: *** [quicklogic/pp3/tests/quicklogic_testsuite/test_logic_cell/CMakeFiles/test_logic_cell-ql-chandalar_analysis.dir/rule] Error 2

[counter_8bit.zip](https://github.com/SymbiFlow/symbiflow-arch-defs/files/4912646/counter_8bit.zip)

rakeshm75, 2020-07-14:
Added one more test case

[test_logic_cell.zip](https://github.com/SymbiFlow/symbiflow-arch-defs/files/4916930/test_logic_cell.zip)

mkurc-ant, 2020-07-14:
Looks like a VPR segfautl. Will check that.

mkurc-ant, 2020-07-16:
I've identified the issue. It appears that i has already been fixed in upstream VPR (as of `aa042b61c4`). I'll go through the procedure of integration upstream VPR into symbiflow. Once that is done a new conda package with the updated VPR will be uploaded with the issue fixed.

mkurc-ant, 2020-07-22:
@rakeshm75 @tpagarani If you cherry pick the following commits from the upstream VTR into your fork then the issue should be solved:  `0f1c496` and `197d65a`.