f4pga/f4pga-arch-defs #1593: Double check system ninja verse conda ninja interaction

mithro, 2020-07-15:
FYI, I removed ninja because a system ninja and the conda ninja interact poorly.  As a follow up, we should figure out a good way to have people enter the conda environment.  My thinking was to put shell scripts in <root>/build like `activate` and `deactivate`.

_Originally posted by @litghost in https://github.com/SymbiFlow/symbiflow-arch-defs/diffs_

mithro, 2020-07-15:
`make enter` should get you a new shell with the conda environment running.

mithro, 2020-07-15:
I also suggest we set up a "Makefile forward" that sends any `make XXX` in the top level to a `cd build/; $(IN_CONDA_ENV) make`?

mithro, 2020-07-15:
@litghost Thoughts?

litghost, 2020-07-15:
> I also suggest we set up a "Makefile forward" that sends any `make XXX` in the top level to a `cd build/; $(IN_CONDA_ENV) make`?

The old forward didn't appear to work with the "env" target, and lead to an infinite loop.  Not sure how it can be done.