f4pga/f4pga-arch-defs #1622: The Clean Circuit step in VPR takes too much time

acomodi, 2020-07-28:
There probably has been a regression in the cleaning circuit step (unsure at the moment which changes caused the regression, if it was VPR or Yosys).

We need to identify what causes this increased time in cleaning the circuit and solve that.

This is crucial for big designs as it impacts every step of the P&R flow (e.g. baselitex P&R takes ~40 seconds at each pack, place and route step).

andrewb1999, 2020-09-04:
@acomodi Has anyone started looking at this issue yet?  On some of my large designs I am seeing the clean circuit step taking 100+ seconds, which severely cuts into the overall p&r speedup I am trying to achieve.

acomodi, 2020-09-04:
@andrewb1999 There was a PR by @litghost that resulted in a great speed-up of the cleaning step in VTR upstream: https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/1477.

We still need to integrate the upstream VTR with the downstream master+wip, so you may want to temporarily cherry-pick those changes until the new master+wip gets in place

litghost, 2020-10-29:
@andrewb1999 The clean circuit step improvement is in.  Has the fix been good enough?