f4pga/f4pga-arch-defs #1625: Feature DLUT.RAM not emitted

mkurc-ant, 2020-07-30:
This issue emerged in the PR https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1502 (SHA `c6a66c0`) on Murax built for A200T (test `tests/9-soc/murax`, target `murax_nexys_video_100_bit_v`).

While decoding the bitstream fasm2bels throws an assertion error meaning that there are LUT RAMs in a SLICEM enabled, but none of them is packed into DLUT hence the `DLUT.RAM` feature is not set and presumably it should be:
```
Traceback (most recent call last):
  File "/tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7a200t-vendor/env/conda/envs/symbiflow_arch_def_base/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7a200t-vendor/env/conda/envs/symbiflow_arch_def_base/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7a200t-vendor/third_party/symbiflow-xc-fasm2bels/fasm2bels/__main__.py", line 4, in <module>
    main()
  File "/tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7a200t-vendor/third_party/symbiflow-xc-fasm2bels/fasm2bels/fasm2bels.py", line 416, in main
    process_tile(top, tile, tile_features)
  File "/tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7a200t-vendor/third_party/symbiflow-xc-fasm2bels/fasm2bels/fasm2bels.py", line 103, in process_tile
    PROCESS_TILE[tile_type](top.conn, top, tile, tile_features)
  File "/tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7a200t-vendor/third_party/symbiflow-xc-fasm2bels/fasm2bels/models/clb_models.py", line 1218, in process_clb
    process_slice(top, slices[s])
  File "/tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7a200t-vendor/third_party/symbiflow-xc-fasm2bels/fasm2bels/models/clb_models.py", line 528, in process_slice
    assert not site.has_feature('{}LUT.RAM'.format(row))
AssertionError
```

Now, the comment in the architecture definition file for SLICEM says that whenever any LUT RAM is used in a SLICEM the `DLUT.RAM` feature should be set. But that's done only for the `DRAM128` mode that uses the whole SLICEM, not for any other. The comment:
```
      <!-- The DLUT must be in RAM-mode for any of the RAM's to work.
           As a corollary, a DRAM requires the clock, so only turn on the
           DRAM if the clock is also connected.
      -->
```

So in the end VPR packs 2 LUT RAMs into ALUT and BLUT thus the `DLUT.RAM` feature is not set. It it unclear whether this behavior is correct or not (haven't check if the design works on HW). Configuration of the particular SLICEM (as emitted by VPR) looks like that:
```
CLBLM_L_X60Y118.SLICEM_X0.AOUTMUX.A5Q
CLBLM_L_X60Y118.SLICEM_X0.COUTMUX.O5
CLBLM_L_X60Y118.SLICEM_X0.FFSYNC
CLBLM_L_X60Y118.SLICEM_X0.DFFMUX.DX
CLBLM_L_X60Y118.SLICEM_X0.A5FFMUX.IN_B
CLBLM_L_X60Y118.SLICEM_X0.A5FF.ZRST
CLBLM_L_X60Y118.SLICEM_X0.A5FF.ZINI=1'b0
CLBLM_L_X60Y118.SLICEM_X0.DFF.ZRST
CLBLM_L_X60Y118.SLICEM_X0.DFF.ZINI=1'b0
CLBLM_L_X60Y118.SLICEM_X0.CLUT.INIT[31:0]=32'b11001100000000001100110000000000
CLBLM_L_X60Y118.SLICEM_X0.CLUT.INIT[63:32]=32'b10000000100000001111111111111111
CLBLM_L_X60Y118.SLICEM_X0.BLUT.DI1MUX.BI
CLBLM_L_X60Y118.SLICEM_X0.BLUT.RAM
CLBLM_L_X60Y118.SLICEM_X0.BLUT.INIT[63:0]=64'b0000000000000000000000000000000000000000000000000000000000000000
CLBLM_L_X60Y118.SLICEM_X0.ALUT.DI1MUX.BDI1_BMC31
CLBLM_L_X60Y118.SLICEM_X0.ALUT.RAM
CLBLM_L_X60Y118.SLICEM_X0.ALUT.INIT[63:0]=64'b0000000000000000000000000000000000000000000000000000000000000000
CLBLM_L_X60Y118.SLICEM_X0.WEMUX.CE
```

litghost, 2020-07-31:
I think we can just add the fasm_mux from https://github.com/SymbiFlow/symbiflow-arch-defs/blob/f0e7b4212544e1d55da776fb7a2ff79117e01454/xc/common/primitives/slicem/slicem.pb_type.xml#L847-L855 and comment to the 
"DRAMs" mode, and this should be fixed.  It isn't an ideal packing, but it should be legal.  @mkurc-ant / @acomodi Could you test this?  I won't be able to get to this until next Tuesday.

litghost, 2020-09-25:
Fixed in #1629

closed 2020-09-25 by litghost