f4pga/f4pga-arch-defs #1636: Support multi clock region ROIs

andrewb1999, 2020-08-12:
After some experimentation with placing a synth tile at the output of a BUFG to try to expose a global clock tree, it seems like this method will not work out of the box.  Placing a synth tile at the wire `CLK_HROW_TOP_R_X87Y78/CLK_HROW_CK_BUFHCLK_R2` causes the clock to be unreachable from anywhere in the ROI.  Seems there needs to be some more explicit emission of the global clock tree or a change in approach.

Also, in some cases there seems to be an issue with constant networks routing in these multi clock region ROIs.  Still trying to narrow this issue down to have a specific test case that causes constant networks to be unable to route.

andrewb1999, 2020-08-12:
@litghost Any suggestions on global clock tree emission?

litghost, 2020-08-12:
So there were two parts to the suggestion.  The first was the synth tile on the wire.  But the second part is to make sure that the graph limits includes portion of that graph.  Did you extend the graph to cover the spine?

andrewb1999, 2020-08-12:
> So there were two parts to the suggestion. The first was the synth tile on the wire. But the second part is to make sure that the graph limits includes portion of that graph. Did you extend the graph to cover the spine?

Oh I misunderstood, I will try that.