f4pga/f4pga-arch-defs #1650: Some synthetic IOs unroutable in overlay architecture

andrewb1999, 2020-08-26:
In the new switch_processing test case provided in #1649 some synthetic IOs are able to be routed and others are not.  With the provided [verilog](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/79200e34ba56aee1f0548451e4884e0306285d49/xc/xc7/tests/switch_processing/switch_processing_arty_overlay.v) for the test case the first error I run into is 
```
Warning 72: No routing path for connection to sink_rr 1580874, retrying with full device bounding box                                                                                                         
Cannot route from BLK-TL-LIOPAD_S[0].IOI_ILOGIC1_O[0] (RR node: 34962 type: SOURCE location: (2,29) class: 57 capacity: 1) to SYN-IOPAD-9[0].outpad[0] (RR node: 1580874 type: SINK location: (11,108) class: 
1 capacity: 1) -- no possible path                                                                                                                                                                            
Failed to route connection from 'sw[2]' to 'in_obuf_pr2_2' for net '$abc$1742$iopadmap$sw' (#8)                                                                                                              
Routing failed.   
```
although, I can run into other errors and even get it to route if I comment out enough not working synth IOs.

Here are the [synth_tiles](https://github.com/andrewb1999/summer-of-code-resources/blob/master/8_26_20/synth_tiles.json) and [architecture](https://github.com/andrewb1999/summer-of-code-resources/blob/master/8_26_20/arch.timing.xml) currently being generated.  The test case can be run locally by cloning #1649 and running the switch_processing_arty_overlay_bit test case.

andrewb1999, 2020-08-26:
@litghost Let me know what other information would be useful

litghost, 2020-08-26:
Can you provide the relevant Vivado wire names for that pair?

andrewb1999, 2020-08-26:
INT_L_X22Y46/WW4A3 and CLBLM_L_X22Y46/CLBLM_WW4B3

andrewb1999, 2020-08-26:
Updated, previously one of the wires was wrong.

andrewb1999, 2020-08-27:
@litghost Here is a route_diag example from an synth inpad trying to reach an LED real IO:
```
Build Device Grid took 0.03 seconds (max_rss 38.5 MiB, delta_rss +0.4 MiB)
Loading routing resource graph
Loading routing resource graph took 18.87 seconds (max_rss 2206.7 MiB, delta_rss +2168.2 MiB)
  RR Graph Nodes: 2014137
  RR Graph Edges: 15022506
  Adding node  1254115 to heap from init route tree with cost 0 (RR node: 1254115 type: SOURCE location: (54,108) class: 1 capacity: 1)
  Routing to 953957 as normal net (BB: 0,0 x 118,163)
  Popping node 1254115 (cost: 0)
    Better cost to 1254115
    New total cost: 0
    New back cost: 0
      Setting path costs for associated node 1254115 (from -1 edge 4294967295)
      Expanding node 1254115 edge 9581588 -> 1254119
      Expanding to node 1254119 (RR node: 1254119 type: OPIN location: (54,108) pin: 1 pin_name: SYN-IOPAD-8[1].inpad[0] capacity: 1)
  Popping node 1254119 (cost: 1.55505e-12)
    Better cost to 1254119
    New total cost: 1.55505e-12
    New back cost: 1.55505e-12
      Setting path costs for associated node 1254119 (from 1254115 edge 9581588)
      Expanding node 1254119 edge 9581590 -> 1255028
      Expanding to node 1255028 (RR node: 1255028 type: CHANY location: (53,106) <-> (53,108) track: 17 len: 2 longline: 0 seg_type: SW2BEG dir: BI_DIR capacity: 1)
Requested lookahead from node 1255028 to 954037
Lookahead returned SW2BEG (29) to IMUX (0) with distance (-57, 0)
Lookahead delay: 8.97e-10
Lookahead congestion: 1.58537e-08
Criticality: 0.99
Lookahead cost: 1.04657e-09
Site pin delay: 0
  Popping node 1255028 (cost: 1.262e-09)
    Better cost to 1255028
    New total cost: 1.262e-09
    New back cost: 6.11698e-12
      Setting path costs for associated node 1255028 (from 1254119 edge 9581590)
      Pruned expansion of node 1255028 edge 9586366 -> 1254121 (to node is IPIN at 54,108x54,108 which does not lead to target block 114,106x114,106)
      Expanding node 1255028 edge 9586367 -> 1253879
      Expanding to node 1253879 (RR node: 1253879 type: CHANX location: (51,107) <-> (54,107) track: 29 len: 3 longline: 0 seg_type: SW2BEG dir: BI_DIR capacity: 1)
Requested lookahead from node 1253879 to 954037
Lookahead returned SW2BEG (29) to IMUX (0) with distance (-57, 0)
Lookahead delay: 8.97e-10
Lookahead congestion: 1.58537e-08
Criticality: 0.99
Lookahead cost: 1.04657e-09
Site pin delay: 0
  Popping node 1253879 (cost: 1.262e-09)
    Better cost to 1253879
    New total cost: 1.262e-09
    New back cost: 6.11698e-12
      Setting path costs for associated node 1253879 (from 1255028 edge 9586367)
      Expanding node 1253879 edge 9580844 -> 1255028
      Expanding to node 1255028 (RR node: 1255028 type: CHANY location: (53,106) <-> (53,108) track: 17 len: 2 longline: 0 seg_type: SW2BEG dir: BI_DIR capacity: 1)
Requested lookahead from node 1255028 to 954037
Lookahead returned SW2BEG (29) to IMUX (0) with distance (-57, 0)
Lookahead delay: 8.97e-10
Lookahead congestion: 1.58537e-08
Criticality: 0.99
Lookahead cost: 1.04657e-09
Site pin delay: 0
  Empty heap (no path found)
Warning 64: No routing path for connection to sink_rr 953957, retrying with full device bounding box
  Adding node  1254115 to heap from init route tree with cost 0 (RR node: 1254115 type: SOURCE location: (54,108) class: 1 capacity: 1)
  Popping node 1254115 (cost: 0)
    Better cost to 1254115
    New total cost: 0
    New back cost: 0
      Setting path costs for associated node 1254115 (from -1 edge 4294967295)
      Expanding node 1254115 edge 9581588 -> 1254119
      Expanding to node 1254119 (RR node: 1254119 type: OPIN location: (54,108) pin: 1 pin_name: SYN-IOPAD-8[1].inpad[0] capacity: 1)
  Popping node 1254119 (cost: 1.55505e-12)
[... 22 lines cut ...]
Lookahead returned SW2BEG (29) to IMUX (0) with distance (-57, 0)
Lookahead delay: 8.97e-10
Lookahead congestion: 1.58537e-08
Criticality: 0.99
Lookahead cost: 1.04657e-09
Site pin delay: 0
  Popping node 1253879 (cost: 1.262e-09)
    Better cost to 1253879
    New total cost: 1.262e-09
    New back cost: 6.11698e-12
      Setting path costs for associated node 1253879 (from 1255028 edge 9586367)
      Expanding node 1253879 edge 9580844 -> 1255028
      Expanding to node 1255028 (RR node: 1255028 type: CHANY location: (53,106) <-> (53,108) track: 17 len: 2 longline: 0 seg_type: SW2BEG dir: BI_DIR capacity: 1)
Requested lookahead from node 1255028 to 954037
Lookahead returned SW2BEG (29) to IMUX (0) with distance (-57, 0)
Lookahead delay: 8.97e-10
Lookahead congestion: 1.58537e-08
Criticality: 0.99
Lookahead cost: 1.04657e-09
Site pin delay: 0
  Empty heap (no path found)
Cannot route from SYN-IOPAD-8[1].inpad[0] (RR node: 1254115 type: SOURCE location: (54,108) class: 1 capacity: 1) to BLK-TL-RIOPAD_SING[0].IOI_OLOGIC0_D1[0] (RR node: 953957 type: SINK location: (114,106) class: 21 capacity: 1) -- no possible path
Routed failed
```
You can see that the synth IO is actually connected to at least a few local rr nodes, but doesn't seem to be able to reach outside of a small local area.  I'm wondering if there is some off by one error with importing switches on the east side of the ROI.

andrewb1999, 2020-08-28:
Okay I have figured out the issue.  Turns out I need to be more careful about where the node is driven from.  I accidentally picked nodes that were going in the wrong direction for the I/O I wanted to use it for. A check for this should be added after importing node names into the database.

closed 2020-08-28 by andrewb1999