f4pga/f4pga-arch-defs #1654: Warnings in make env

SSandeep19, 2020-09-02:
Hi,

I am a Phd student working on FPGA security. I am trying to use Symbiflow to generate bitstream for Xilinx Artix-7.  I am facing multiple problems while installing the toolchain by following the steps given in getting started section from symbiflow (https://symbiflow.readthedocs.io/en/latest/symbiflow-arch-defs/docs/source/getting-started.html). I have cloned the repository from github and when i executed **"make env"** in symbiflow-arch-defs there are lot of warnings related to vivado as shown in the attached screenshot. There is another error also regarding Sanitizers which i am attaching the screenshot for. I want to know is this the right way to install the tools and please let me know if any changes are required in my approach.
My goal is to generate bitstream for a given verilog file using symbiflow for Artix 7 FPGA. I am working in Ubuntu without vivado installed in the system.
Any help would be greatly appreciated. Please point me in the right direction if this is not the place to ask.

[image]
[image]

Regards,
Sandeep
ss1548@wildcats.unh.edu

SSandeep19, 2020-09-02:
Yes i actually tried to install Vivado for linux and the download constantly failed. I want to know if that is what the warnings are about and is Vivado needed for symbiflow to work or is there anything else to the warnings. I couldn't find any packages for Sanitizers.

litghost, 2020-09-02:
Vivado is not required.  Targets that invoke Vivado (e.g. the targets printed in the warnings) will not work without it.  If you aren't invoking one of those targets, I don't believe you care.

SSandeep19, 2020-09-02:
Will those targets be invoked if i am generating a bitstream for Xilinx Artix-7 ?. If i follow the next step provided by symbiflow in their website (https://symbiflow.readthedocs.io/en/latest/symbiflow-arch-defs/docs/source/getting-started.html). In the documentation it is asking to execute make all_conda but it is throwing another error.
[image]

litghost, 2020-09-02:
> Will those targets be invoked if i am generating a bitstream for Xilinx Artix-7 ?

No

> If i follow the next step provided by symbiflow in their website (https://symbiflow.readthedocs.io/en/latest/symbiflow-arch-defs/docs/source/getting-started.html). In the documentation it is asking to execute make all_conda but it is throwing another error.

That instruction is out of date, it is no longer required.

HackerFoo, 2020-09-02:
Hi Sandeep,

There is also an alternative way to build Symbiflow packages that may be easier for you, especially if have used Nix: https://github.com/hackerfoo/nix-symbiflow

You'll need to change `use-vivado ? true` to `use-vivado ? false` on the third line of `default.nix`.

Then:
```
nix-build ~/src/nix-symbiflow -A symbiflow-arch-defs
```

will build the toolchain on your machine.

Also note that https://github.com/SymbiFlow/symbiflow-examples will download a prebuilt toolchain, if you'd rather not wait several hours to build the toolchain.

-- Dusty

SSandeep19, 2020-09-02:
Hi litghost,

Is there any updated document i can refer to ? Does it mean symbiflow is build and ready to use ? if so what are the commands i should run next ?

litghost, 2020-09-02:
> Hi litghost,
> 
> Is there any updated document i can refer to ? Does it mean symbiflow is build and ready to use ? if so what are the commands i should run next ?

There is an example directly below in the `getting-started` you linked?  There is also instructions on https://github.com/SymbiFlow/symbiflow-examples

HackerFoo, 2020-09-03:
@SSandeep19 While the Nix packages can work for users, I recommend trying https://github.com/SymbiFlow/symbiflow-examples first, as the Nix packages track current development and will take a long time to build.

SSandeep19, 2020-09-03:
Hi litghost,

I tried the example provided in getting started document and i faced an error, attached screenshot of it. I want to know how to generate bitstream of the verilog file i have using symbiflow please provide if there is any documentation as such, i tried to find the verilog file of the counter provided as example and couldn't find it.

acomodi, 2020-09-03:
Hi @SSandeep19, this error is a result of CMake not emitting the `<test>_bin` targets any more due to a recent change. The readme does not reflect this change. You can generate the bitstream using the following target:

```
make counter_arty_bit
```

If your aim though is just to generate bitstreams from a verilog design, without experimenting on the architecture itself, it might be better to make use of the [symbiflow examples](https://github.com/SymbiFlow/symbiflow-examples) repository as suggested above.

SSandeep19, 2020-09-03:
Hi acomodi,

Thank you it worked. I actually need to work on the toolchain architecture like how to just run Yosis and get eblif file or just run VPR. Is there a way to do it and how to run my verilog file because i cannot figure out where to input my verilog file. Please let me know if there is any documentation. Do you still think https://github.com/SymbiFlow/symbiflow-examples will serve my purpose better.

SSandeep19, 2020-09-03:
Hi tcal-x,

Thank you for your reply. I wanted to work on the tool chain architecture like to run yosys to get eblif file or give eblif file as input and run VPR to get net,place and route files. I also want to generate bit file for my verilog code not just the examples given in symbiflow. Any documentation or suggestions on this would greatly help me. Or do you think i can get this done by symbiflow examples.

tcal-x, 2020-09-03:
@SSandeep19 , I agree with the others that symbiflow-examples is the best place to start.   It has a few examples for Artix 7-series devices, and the makefile structure is straightforward for copying and creating your own project.

SSandeep19, 2020-09-04:
Hi tcal-x,
Thank you and i started with SymbiFlow examples and i am facing some errors with it so i have a raised a issue in that project and this is the link for the issue https://github.com/SymbiFlow/symbiflow-examples/issues/45#issue-693115212. Any help would be very much appreciated.

litghost, 2020-10-29:
@SSandeep19 I believe this issue is resolved.  Please let us know if there is still something remaining.

closed 2020-10-29 by litghost