f4pga/f4pga-arch-defs #1661: [URay] Problematic XIPHY_BYTE_RIGHT/XIPHY_BITSLICE_TILE_57_RX_D wire

mkurc-ant, 2020-09-11:
This wire fails with the current tile pin direction assignment procedure.

In some `XIPHY_BYTE_RIGHT` tile instances this wire belongs to a node that goes only to `HPIO_RIGHT/HPIOB_M` site. This is fine. But for others the same wire also connects to the interconnect of `CMT_RIGHT` tile.

Since tile pin directions are assigned per tile type, not per tile instance, there is a conflict in node classification that leads to an assertion failure.

litghost, 2020-09-11:
There should be a consistent pin direction, but the current algorithm might not be good enough to find it.  One thing to try is see if we can extract pin direction based on ordered wires from nodes, rather than the current approach.

mkurc-ant, 2020-09-14:
Ok, I can try that approach. Meanwhile I limited the range of the grid being imported to clock regions X0Y1 and X0Y2 only.