f4pga/f4pga-arch-defs #1670: Kokoro disk is full

mkurc-ant, 2020-09-22:
While trying to investigate what I've broken in https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1651 I stumbled upon "disk full" messages like:
```
sqlite3.OperationalError: database or disk is full
```
```
capnp.lib.capnp.KjException: kj/io.c++:406: overloaded: ::writev(fd, current, iov.end() - current): No space left on device; fd = 10
```
Found these in "Xilinx Series 7 - Vendor Tool Tests (Presubmit)" logs.

litghost, 2020-10-29:
I've seen this too.  I created https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1725 to debug this, and it looks like the kokoro working directory is a magic 4 TB mount point.  Presumably this underlying magic may fail, which manifests as disk full?  Unclear what is actually happening here.  Maybe a GCP bug?

litghost, 2020-12-04:
`/home/kbuilder/.pyenv/libexec/pyenv-init: line 131: cannot create temp file for here-document: No space left on device`

Looks like `HOME` might be full?

litghost, 2020-12-04:
Example failure:
```
Filesystem      Size  Used Avail Use% Mounted on
udev             52G     0   52G   0% /dev
tmpfs            11G   17M   11G   1% /run
/dev/sda1        99G   95G     0 100% /
tmpfs            52G     0   52G   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs            52G     0   52G   0% /sys/fs/cgroup
/dev/sdc1       246G   43G  203G  18% /opt/Xilinx
cgmfs           100K     0  100K   0% /run/cgmanager/fs
tmpfs            11G     0   11G   0% /run/user/1000
/dev/sdd        246G   29G  217G  12% /image
```

PR #1847 (2020-12-10): Move HOME to /tmpfs.
Fixes #1670
closed 2020-12-10 by litghost