f4pga/f4pga-arch-defs #1684: Support BRAM_Rs

andrewb1999, 2020-09-30:
Is there anything major holding back enabling BRAM_Rs? Losing half of the BRAMs available on a chip has become a significant limitation of how large the designs I can map through symbiflow are.

I am willing to put in some time to add this support, but I'm not entirely sure why they are disabled in the first place.

andrewb1999, 2020-09-30:
@litghost Can you give some advice on this when you have a chance?

litghost, 2020-09-30:
This is actually pretty straight forward to fix, but it will require some fixes and a bunch of testing.

1.  Need to properly fix https://github.com/SymbiFlow/prjxray/issues/1285 . @tmichalak has proposed a couple fixes, but it is unclear if the fixes are generally safe and structurely true to the underlying bitstream.
2. Update symbiflow-arch-defs to add the BRAM_R.
 - Make sure https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/xc/common/primitives/bram_r is in sync with https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/xc/common/primitives/bram_l
 - https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc/xc7/archs/artix7/devices/xc7a50t-virt/CMakeLists.txt and equivalent to add BRAM_R
 - Add cluster equivalance to https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc/xc7/archs/artix7/tiles/bram_r/CMakeLists.txt and https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc/xc7/archs/artix7/tiles/bram_l/CMakeLists.txt
 - Add tests confirming that BRAM_L clusters can be placed in BRAM_R and vise versa
 - Add tests filling all BRAM's in the fabric and checking the init in hardware, and via fasm2bels, etc

andrewb1999, 2020-09-30:
> This is actually pretty straight forward to fix, but it will require some fixes and a bunch of testing.
> 
>     1. Need to properly fix [SymbiFlow/prjxray#1285](https://github.com/SymbiFlow/prjxray/issues/1285) . @tmichalak has proposed a couple fixes, but it is unclear if the fixes are generally safe and structurely true to the underlying bitstream.
> 
>     2. Update symbiflow-arch-defs to add the BRAM_R.
> 
> 
>     * Make sure https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/xc/common/primitives/bram_r is in sync with https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/xc/common/primitives/bram_l
> 
>     * Update https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc/xc7/archs/artix7/devices/xc7a50t-virt/CMakeLists.txt and equivalent devices to add BRAM_R
> 
>     * Add cluster equivalence to https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc/xc7/archs/artix7/tiles/bram_r/CMakeLists.txt and https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc/xc7/archs/artix7/tiles/bram_l/CMakeLists.txt
> 
>     * Add tests confirming that BRAM_L clusters can be placed in BRAM_R and vise versa
> 
>     * Add tests filling all BRAM's in the fabric and checking the init in hardware, and via fasm2bels, etc
> 
> 
> > [image][image] GitHub**[SymbiFlow/symbiflow-arch-defs](https://github.com/SymbiFlow/symbiflow-arch-defs)**FOSS architecture definitions of FPGA hardware useful for doing PnR device generation. - SymbiFlow/symbiflow-arch-defs
> 
> > [image][image] GitHub**[SymbiFlow/symbiflow-arch-defs](https://github.com/SymbiFlow/symbiflow-arch-defs)**FOSS architecture definitions of FPGA hardware useful for doing PnR device generation. - SymbiFlow/symbiflow-arch-defs
> 
> > [image][image] GitHub**[SymbiFlow/symbiflow-arch-defs](https://github.com/SymbiFlow/symbiflow-arch-defs)**FOSS architecture definitions of FPGA hardware useful for doing PnR device generation. - SymbiFlow/symbiflow-arch-defs
> 
> > [image][image] GitHub**[SymbiFlow/symbiflow-arch-defs](https://github.com/SymbiFlow/symbiflow-arch-defs)**FOSS architecture definitions of FPGA hardware useful for doing PnR device generation. - SymbiFlow/symbiflow-arch-defs
> 
> > [image][image] GitHub**[SymbiFlow/symbiflow-arch-defs](https://github.com/SymbiFlow/symbiflow-arch-defs)**FOSS architecture definitions of FPGA hardware useful for doing PnR device generation. - SymbiFlow/symbiflow-arch-defs

Sounds good.  How can we determine if @tmichalak 's fixes are correct?  Testing?