f4pga/f4pga-arch-defs #1685: Reduce lookahead creation time for partition regions

andrewb1999, 2020-09-30:
Currently the slowest part of defining and testing designs with partition regions is the time it takes to generate the place delay and router lookahead.  I am wondering if there is some way to get a good enough approximation of these files by creating them from pre-generated full chip files rather than generating them from scratch for each new partition region.

A flow like this would allow faster creation of different partitioned designs and more design space exploration opportunities.

andrewb1999, 2020-09-30:
@acomodi I'm guessing you have the most experience with this given the extended map work.  Is this something you think is feasible?

acomodi, 2020-10-01:
@andrewb1999 I believe that using the full device lookahead file could be done. The only problem I would see in there is the fact that, depending on the architecture file, the segment indexes could have an offset w.r.t. the partition region architectures.
This causes inconsistencies when querying the lookahead as it gets addressed by segment_index.

For instance, if a partition region lacks some of the segments present in the whole chip (e.g. clock tile related wires), the whole-device lookahead would be unusable.

Instead, if there is a prior mapping of segment indices between the partition region architecture and the whole-chip architecture, I believe that the lookahead could be re-built based on that mapping and used for the partition region.
I am unsure about special wires, but in general, the lookahead returns an expected cost based on the distance to the destination from a given location. Given that the partition region would be a subset of the whole device, all the distances for the pr should be within the whole-device lookahead and should return valid costs.

I do not have much experience on the delay matrix though, but I think that a subset of the delay matrix corresponding to the partition region grid could be extracted and used as well.

litghost, 2020-10-01:
The delay matrix should be generalizable, because it's totally flat.  I think the extensions that @acomodi has in mind might work.  It would be super interesting to see if a A100T lookahead worked well on the A50T and the A200T fabrics too.