f4pga/f4pga-arch-defs #1721: Make env error: json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

RetownPlato, 2020-10-24:
After clone the repo, I ran `make env`.
```
==> WARNING: A newer version of conda exists. <==
  current version: 4.8.3
  latest version: 4.9.0

Please update conda by running

    $ conda update -n base -c defaults conda

Preparing transaction: failed

# >>>>>>>>>>>>>>>>>>>>>> ERROR REPORT <<<<<<<<<<<<<<<<<<<<<<

    Traceback (most recent call last):
      File "/home/allen/symbiflow-arch-defs/env/conda/lib/python3.8/site-packages/conda/exceptions.py", line 1079, in __call__
        return func(*args, **kwargs)
      File "/home/allen/symbiflow-arch-defs/env/conda/lib/python3.8/site-packages/conda_env/cli/main.py", line 80, in do_call
        exit_code = getattr(module, func_name)(args, parser)
      File "/home/allen/symbiflow-arch-defs/env/conda/lib/python3.8/site-packages/conda_env/cli/main_update.py", line 124, in execute
        result[installer_type] = installer.install(prefix, specs, args, env)
      File "/home/allen/symbiflow-arch-defs/env/conda/lib/python3.8/site-packages/conda_env/installers/conda.py", line 40, in install
        unlink_link_transaction.execute()
      File "/home/allen/symbiflow-arch-defs/env/conda/lib/python3.8/site-packages/conda/core/link.py", line 244, in execute
        self.verify()
      File "/home/allen/symbiflow-arch-defs/env/conda/lib/python3.8/site-packages/conda/common/io.py", line 88, in decorated
        return f(*args, **kwds)
      File "/home/allen/symbiflow-arch-defs/env/conda/lib/python3.8/site-packages/conda/core/link.py", line 221, in verify
        self.prepare()
      File "/home/allen/symbiflow-arch-defs/env/conda/lib/python3.8/site-packages/conda/core/link.py", line 210, in prepare
        grps = self._prepare(self.transaction_context, stp.target_prefix,
      File "/home/allen/symbiflow-arch-defs/env/conda/lib/python3.8/site-packages/conda/core/link.py", line 287, in _prepare
        packages_info_to_link = tuple(read_package_info(prec, pcrec)
      File "/home/allen/symbiflow-arch-defs/env/conda/lib/python3.8/site-packages/conda/core/link.py", line 287, in <genexpr>
        packages_info_to_link = tuple(read_package_info(prec, pcrec)
      File "/home/allen/symbiflow-arch-defs/env/conda/lib/python3.8/site-packages/conda/gateways/disk/read.py", line 100, in read_package_info
        paths_data = read_paths_json(epd)
      File "/home/allen/symbiflow-arch-defs/env/conda/lib/python3.8/site-packages/conda/gateways/disk/read.py", line 170, in read_paths_json
        data = json.load(paths_json)
      File "/home/allen/symbiflow-arch-defs/env/conda/lib/python3.8/json/__init__.py", line 293, in load
        return loads(fp.read(),
      File "/home/allen/symbiflow-arch-defs/env/conda/lib/python3.8/json/__init__.py", line 357, in loads
        return _default_decoder.decode(s)
      File "/home/allen/symbiflow-arch-defs/env/conda/lib/python3.8/json/decoder.py", line 337, in decode
        obj, end = self.raw_decode(s, idx=_w(s, 0).end())
      File "/home/allen/symbiflow-arch-defs/env/conda/lib/python3.8/json/decoder.py", line 355, in raw_decode
        raise JSONDecodeError("Expecting value", s, err.value) from None
    json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

`$ /home/allen/symbiflow-arch-defs/env/conda/bin/conda-env update --name symbiflow_arch_def_base --file environment.yml`

  environment variables:
                 CIO_TEST=<not set>
         CONDA_ALWAYS_YES=1
  CONDA_AUTO_UPDATE_CONDA=false
        CONDA_DEFAULT_ENV=base
                CONDA_EXE=/home/allen/symbiflow-arch-defs/env/conda/bin/conda
             CONDA_PREFIX=/home/allen/symbiflow-arch-defs/env/conda
    CONDA_PROMPT_MODIFIER=(base)
         CONDA_PYTHON_EXE=/home/allen/symbiflow-arch-defs/env/conda/bin/python
               CONDA_ROOT=/home/allen/symbiflow-arch-defs/env/conda
              CONDA_SHLVL=1
            DEFAULTS_PATH=/usr/share/gconf/ubuntu.default.path
                 INFOPATH=/home/linuxbrew/.linuxbrew/share/info:
          LD_LIBRARY_PATH=/lib:/usr/lib:/usr/local/lib:/home/linuxbrew/.linuxbrew/lib
           MANDATORY_PATH=/usr/share/gconf/ubuntu.mandatory.path
                  MANPATH=/home/linuxbrew/.linuxbrew/share/man::
                     PATH=/home/allen/symbiflow-arch-defs/env/conda/bin:/home/allen/symbiflow-
                          arch-defs/env/conda/bin:/home/allen/symbiflow-arch-defs/env/conda/cond
                          abin:/home/linuxbrew/.linuxbrew/bin:/home/linuxbrew/.linuxbrew/sbin:/o
                          pt/Xilinx/DocNav:/opt/Xilinx/Vivado/2020.1/bin:/home/allen/intelFPGA_l
                          ite/20.1/quartus/bin:/home/allen/bin:/usr/local/bin:/home/allen/.local
                          /bin:/home/allen/.local/bin:/home/allen/bin:/usr/local/sbin:/usr/local
                          /bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/b
                          in
       REQUESTS_CA_BUNDLE=<not set>
            SSL_CERT_FILE=<not set>
               WINDOWPATH=2

     active environment : base
    active env location : /home/allen/symbiflow-arch-defs/env/conda
            shell level : 1
       user config file : /home/allen/.condarc
 populated config files : /home/allen/symbiflow-arch-defs/env/conda/.condarc
          conda version : 4.8.3
    conda-build version : not installed
         python version : 3.8.3.final.0
       virtual packages : __glibc=2.31
       base environment : /home/allen/symbiflow-arch-defs/env/conda  (writable)
           channel URLs : https://repo.anaconda.com/pkgs/main/linux-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/linux-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /home/allen/symbiflow-arch-defs/env/downloads/conda-pkgs
       envs directories : /home/allen/symbiflow-arch-defs/env/conda/envs
                          /home/allen/.conda/envs
               platform : linux-64
             user-agent : conda/4.8.3 requests/2.23.0 CPython/3.8.3 Linux/5.4.22-20183886-1 ubuntu/20.04.1 glibc/2.31
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False

An unexpected error has occurred. Conda has prepared the above report.

Upload successful.
make: *** [third_party/make-env/conda.mk:128: /home/allen/symbiflow-arch-defs/env/conda/envs/symbiflow_ar_def_base/bin/python] Error 1
```

RetownPlato, 2020-10-24:
In third_party/make-env/conda.mk:128
After plug in the variable is:
```
source /home/allen/symbiflow-arch-defs/env/conda/bin/activate && conda env update --name symbiflow_arch_def_base --file environment.yml
```

litghost, 2020-10-26:
I've never seen this before.  Can you please try again with a clean repo today, and see if it replicates?

litghost, 2020-10-29:
@RetownPlato If you can replicate this, please feel free to re-open this.  Closing for now.

closed 2020-10-29 by litghost