f4pga/f4pga-arch-defs #1750: Artix 200T missing bits

acomodi, 2020-11-04:
While adding designs in fpga-tool-perf for the nexys_video board, the following error showed up.

```
File "/media/newdrive/prjxray/fpga-tool-perf/env/conda/envs/symbiflow-env/lib/python3.8/site-packages/prjxray/fasm_assembler.py", line 190, in parse_fasm_filename
    raise FasmLookupError('\n'.join(missing_features))
prjxray.fasm_assembler.FasmLookupError: Segment DB CMT_TOP_L_LOWER_B, key CMT_TOP_L_LOWER_B.MMCM_CLK_FREQ_BB_REBUF3_NS.MMCM_CLK_FREQ_BB_NS3 not found from line 'CMT_TOP_L_LOWER_B_X256Y113.MMCM_CLK_FREQ_BB_REBUF3_NS.MMCM_CLK_FREQ_BB_NS3'
Segment DB CMT_TOP_L_UPPER_T, key CMT_TOP_L_UPPER_T.PLL_CLK_FREQ_BB3_NS.PLL_CLK_FREQ_BB_BUFOUT_NS3 not found from line 'CMT_TOP_L_UPPER_T_X256Y96.PLL_CLK_FREQ_BB3_NS.PLL_CLK_FREQ_BB_BUFOUT_NS3'
Segment DB CMT_TOP_L_LOWER_B, key CMT_TOP_L_LOWER_B.MMCM_CLK_FREQ_BB_REBUF3_NS.MMCM_CLK_FREQ_BB_NS3 not found from line 'CMT_TOP_L_LOWER_B_X256Y61.MMCM_CLK_FREQ_BB_REBUF3_NS.MMCM_CLK_FREQ_BB_NS3'
Segment DB CMT_TOP_L_UPPER_T, key CMT_TOP_L_UPPER_T.PLL_CLK_FREQ_BB3_NS.PLL_CLK_FREQ_BB_BUFOUT_NS3 not found from line 'CMT_TOP_L_UPPER_T_X256Y44.PLL_CLK_FREQ_BB3_NS.PLL_CLK_FREQ_BB_BUFOUT_NS3'
Segment DB GTP_COMMON_MID_RIGHT, key GTP_COMMON_MID_RIGHT.HCLK_GTP_CK_IN2.HCLK_GTP_CK_MUX2 not found from line 'GTP_COMMON_MID_RIGHT_X167Y23.HCLK_GTP_CK_IN2.HCLK_GTP_CK_MUX2'
Segment DB GTP_COMMON_MID_RIGHT, key GTP_COMMON_MID_RIGHT.HCLK_GTP_CK_IN0.HCLK_GTP_CK_MUX0 not found from line 'GTP_COMMON_MID_RIGHT_X167Y23.HCLK_GTP_CK_IN0.HCLK_GTP_CK_MUX0'
Makefile:43: recipe for target 'top.bit' failed
make: *** [top.bit] Error 1
```

The problem, as far as I understood, might be related to the fact that the bottom and top rows of the 200T have a non-common structure, with the `GTP_COMMON_MID` tile in the middle of the row.

Placing the PLL elsewhere, avoiding the top and bottom rows seems to solve the issue.

A similar issue was filed, and I believe we are hitting the same kind of problem here: https://github.com/SymbiFlow/symbiflow-arch-defs/issues/1489

acomodi, 2020-11-04:
The image below is from a dcp obtained with the yosys-vivado toolchain in FPGA-tool-perf. The highlighted nets are the CLKIN and CLKOUTs nets that go to/from the PLL in the X1Y0 cmt.

[image]

The unknown GTP bits are highlighted in the image below, and we are indeed missing those. Given that the routing step passes correctly in VPR, I think that the RR nodes passing through the GTP tile are correctly connected, but we have some missing bits for those pips.
The issue though is that we would need to move the whole fuzzer generation to use the 200T device, as now, IIRC, we are using the 100T.

[image]

mkurc-ant, 2020-11-04:
MMCM tile `REBUF` PIPs are disabled in 034b now. Enabling them should solve at least some of the missing bits.

mithro, 2020-11-13:
@mkurc-ant - What do we need to do to enable the `REBUF` PIPs? Why were they disabled?

mkurc-ant, 2020-11-16:
@mithro I don't know why they are disabled now. A preliminary test I've done shows that the fuzzer solves them successfully. Haven't checked for potential bit conflicts though.

mkurc-ant, 2020-11-17:
This should solve the unknown MMCM PIPs: https://github.com/SymbiFlow/prjxray/pull/1498

acomodi, 2020-11-17:
Perfect. The bits to solve now are related to the pips in the `GTP_COMMON_MID` tile.
Even though the 50T and 100T have the `GTP_COMMON` tiles, their location at the border of the device makes them not interfering with the clock wires.

Instead, for the 200T, the `GTP_COMMON_MID` tiles alter the path of the clock wires which would normally get from/to the PLL to/from the BUFHCE tiles. 

I am trying to run the fuzzers for the 200T, and see if we can transition to that part in prjxray. This would give us the possibility to write a fuzzer to get the missing bits for those pips (visible in the above screenshot)

mkurc-ant, 2020-11-24:
I have a finding. So the `CMT_TOP_L_UPPER_T` PIPs with missing bits are actually bi-directional pseudo PIPs. For example we have an entry in the db for `CMT_TOP_L_UPPER_T.PLL_CLK_FREQ_BB_BUFOUT_NS3.PLL_CLK_FREQ_BB3_NS` but we do not have one for `CMT_TOP_L_UPPER_T.PLL_CLK_FREQ_BB3_NS.PLL_CLK_FREQ_BB_BUFOUT_NS3`.

I'm not entirely sure how we model bi-directional PIPs in VPR.

One solution that comes into my mind is to have two entries per bi-dir PPIP for both directions.